nextcloud/nextcloudpi #1088: build-docker.sh does not work for armhf

mad73923, 2020-02-26:
```
$ ./build-docker.sh armhf    
#2 [internal] load build definition from Dockerfile
#2 transferring dockerfile: 574B done
#2 DONE 0.1s

#1 [internal] load .dockerignore
#1 transferring context: 34B done
#1 DONE 0.1s

#4 [internal] load metadata for docker.io/arm32v7/debian:buster-slim
#4 DONE 1.1s

#3 [internal] load metadata for docker.io/library/debian:buster-slim
#3 DONE 1.1s

#6 [qemu 1/2] FROM docker.io/library/debian:buster-slim@sha256:b137d31f0ebc...
#6 DONE 0.0s

#5 [stage-1 1/4] FROM docker.io/arm32v7/debian:buster-slim@sha256:cd93e4200...
#5 DONE 0.0s

#7 [qemu 2/2] RUN apt-get update && DEBIAN_FRONTEND=noninteractive apt-get ...
#7 CACHED

#8 [stage-1 2/4] COPY --from=qemu /usr/bin/qemu-arm-static /usr/bin/
#8 CACHED

#10 [internal] load build context
#10 transferring context: 115B done
#10 DONE 0.0s

#9 [stage-1 3/4] RUN mkdir -p /etc/services-available.d  /etc/services-enab...
#9 0.554 standard_init_linux.go:211: exec user process caused "exec format error"
#9 ERROR: executor failed running [/bin/sh -c mkdir -p /etc/services-available.d  /etc/services-enabled.d]: runc did not terminate sucessfully
------
 > [stage-1 3/4] RUN mkdir -p /etc/services-available.d  /etc/services-enabled.d:
------
failed to solve with frontend dockerfile.v0: failed to build LLB: executor failed running [/bin/sh -c mkdir -p /etc/services-available.d  /etc/services-enabled.d]: runc did not terminate sucessfully
```

I tried to buid the docker for armhf plattform. With no success.
The images on docker hub are outdated. These should be updated.

nachoparker, 2020-02-27:
You need `qemu-static` and `binfmt` installed on your system to build the images.

Also, you can just ncp-update the current images to get an up to date system, in the meantime while we upload new ones.

Cheers

closed 2020-02-27 by nachoparker

mad73923, 2020-02-28:
Thanks, this did the trick!