nextcloud/nextcloudpi #1095: Nextcloud 18.0.1 Docker image request - two version upgrade not possible

sunjam, 2020-03-03:
Related to #1060, but specifically for ncp Docker images.  Docker pulls version Nextcloud 16.0.3, and *nc-update-nextcloud* fails:

`
Exception: Updates between multiple major versions and downgrades are unsupported.  Update failed`

Thanks for the great software!

nachoparker, 2020-03-03:
yeah, you'll have to upgrade to 17, then to 18, or if it's a new instance just use `nc-nextcloud` + `nc-init` to install directly version 18.

This is a Nextcloud updater limitation

closed 2020-03-03 by nachoparker

ercoppa, 2020-03-07:
@nachoparker I hope you reconsider and treat this as a bug for this project. Indeed, a fresh install of nextcloudpi using the docker image will lead to this situation:
1) docker pulls the current image, which is based on NC 16
2) the user attempts to update using `ncp-config`: after confirming to use "0" (default value) as the release, ncp will try to update to NC18 without asking anything to the user, leading to a failure and breaking NC.
3) ncp will attempt to restore a backup, failing because under docker the filesystem is "overlayfs", while the script supports restore only on "plain" ext/btrfs filesystems.

Two fixes that you consider:
1) ncp could first verify whether the version to update is "one major release" after the installed version: if this is not the case, it should abort (before breaking NC), suggesting to the user to manually update to the major version X where X is the next major release after the current release.
1b) [alternative] ncp could ask for confirmation before proceeding with the update
2) ncp when attempting to restore a backup on the overlayfs could ask to the user if the underlying filesystem is ext. I quickly looked at the script restoring a backup and it should work on the overlayfs. Am I wrong?

If you consider (1) and (2) together, you see that the current scenario for a docker user is far from ideal.

nachoparker, 2020-03-18:
I pushed NC images with 18.0.2 for all architectures