nextcloud/nextcloudpi #1096: Automount failure

Bobo-amg, 2020-03-03:
NextCloudPi v1.21.0

Issue:
Server restart results in MariaDB not starting BECAUSE automount has failed.

Steps to reproduce:
After successful server setup without external drive, documentation was followed to move data and database dir over to USB drive (with automount active). Server restarted but external drive was NOT mounted.

Bobo-amg, 2020-03-07:
Narrowing this down further.

The issue is specifically related to moving the database. Once that is done (and performing a reboot) causes the issue. The external drive is an ssd. 

Relevant log from dmesg as follows:

[Sat Mar  7 08:11:30 2020] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[Sat Mar  7 08:11:30 2020] xhci_hcd 0000:01:00.0: HC died; cleaning up
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#15 uas_eh_abort_handler 0 uas-tag 6 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#15 CDB: opcode=0x2a 2a 00 00 48 b1 38 00 04 00 00
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#14 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#14 CDB: opcode=0x2a 2a 00 00 48 b0 f8 00 00 20 00
[Sat Mar  7 08:11:30 2020] usb 1-1: USB disconnect, device number 2
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#8 uas_eh_abort_handler 0 uas-tag 3 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#8 CDB: opcode=0x2a 2a 00 00 46 a1 38 00 02 00 00
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#5 uas_eh_abort_handler 0 uas-tag 15 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#5 CDB: opcode=0x2a 2a 00 00 48 c1 38 00 00 40 00
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#4 uas_eh_abort_handler 0 uas-tag 9 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#4 CDB: opcode=0x2a 2a 00 00 48 bd 38 00 04 00 00
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#3 uas_eh_abort_handler 0 uas-tag 5 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#3 CDB: opcode=0x28 28 00 00 01 66 80 00 00 20 00
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#2 uas_eh_abort_handler 0 uas-tag 4 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#2 CDB: opcode=0x28 28 00 00 01 c2 c0 00 00 20 00
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#1 uas_eh_abort_handler 0 uas-tag 2 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#1 CDB: opcode=0x2a 2a 00 00 46 9f 38 00 02 00 00
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#0 uas_eh_abort_handler 0 uas-tag 8 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x2a 2a 00 00 48 b9 38 00 04 00 00
[Sat Mar  7 08:11:30 2020] usb 2-1: USB disconnect, device number 2
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#9 uas_zap_pending 0 uas-tag 10 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#9 CDB: opcode=0x28 28 00 00 47 5a f8 00 00 40 00
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#10 uas_zap_pending 0 uas-tag 11 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#10 CDB: opcode=0x28 28 00 00 47 5a c8 00 00 10 00
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#11 uas_zap_pending 0 uas-tag 12 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#11 CDB: opcode=0x28 28 00 00 47 59 78 00 00 10 00
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#12 uas_zap_pending 0 uas-tag 13 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#12 CDB: opcode=0x28 28 00 00 47 58 48 00 00 60 00
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#13 uas_zap_pending 0 uas-tag 14 inflight: CMD
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] tag#13 CDB: opcode=0x28 28 00 00 47 56 98 00 00 10 00
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: Device offlined - not ready after error recovery
[Sat Mar  7 08:11:30 2020] print_req_error: I/O error, dev sda, sector 4676344
[Sat Mar  7 08:11:30 2020] BTRFS error (device sda1): bdev /dev/sda1 errs: wr 0, rd 1, flush 0, corrupt 0,                                    gen 0
[Sat Mar  7 08:11:30 2020] print_req_error: I/O error, dev sda, sector 4676296
[Sat Mar  7 08:11:30 2020] print_req_error: I/O error, dev sda, sector 4675960
[Sat Mar  7 08:11:30 2020] print_req_error: I/O error, dev sda, sector 4675656
[Sat Mar  7 08:11:30 2020] print_req_error: I/O error, dev sda, sector 4675224
[Sat Mar  7 08:11:30 2020] print_req_error: I/O error, dev sda, sector 4764984
[Sat Mar  7 08:11:30 2020] print_req_error: I/O error, dev sda, sector 4763960
[... cut ...]
[Sat Mar  7 08:11:30 2020] BTRFS warning (device sda1): Skipping commit of aborted transaction.
[Sat Mar  7 08:11:30 2020] BTRFS: error (device sda1) in cleanup_transaction:1860: errno=-5 IO failure
[Sat Mar  7 08:11:30 2020] BTRFS info (device sda1): delayed_refs has NO entry
[Sat Mar  7 08:11:30 2020] sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyt                                   e=0x00
[Sat Mar  7 08:11:31 2020] xhci_hcd 0000:01:00.0: WARN Can't disable streams for endpoint 0x82, streams ar                                   e being disabled already

Bobo-amg, 2020-03-07:
Just to add, moving the data over first did not result in the above. ONLY occurs once the database is moved over.

ovpc, 2020-03-10:
> (device sda1): forced readonly

You're drive is somehow failing at boot and mounted readonly, this is most likely not a bug.
Please go forum if need assistance investigating.

Bobo-amg, 2020-03-10:
Yes, that is what I thought. However, i have tested the drive on std raspbian without issue.
**I also tested it with just the data (no database) on the drive, again without issue.**
So I believe this may well be a bug relating to the mariadb database when it starts. Perhaps some sort of timing issue?

ovpc, 2020-03-10:
Not sure but remember, mariadb not waiting for mount of external drive and timing out, 
did happen before, but thought it was fixed by @nachoparker 
So may I suggest try btrfs-check from another pc.

nachoparker, 2020-03-10:
yes, this was fixed long time ago, and I haven't heard of other cases in years... I wonder what happened in this case.

Let's reopen if we hear more cases like this

closed 2020-03-10 by nachoparker

Bobo-amg, 2020-03-11:
Ok, so have done some more testing, and the last piece of the puzzle for me was the BTRFS filesystem.
Instead of formatting the usb drive to BTRFS I formatted it as ext4. Boom! All issues evaporated.

**So there is something wrong with the following combination:
RPi 4
both database and data dir on external usb ssd formatted with BTRFS**

Hope this might help out somebody else.