nextcloud/nextcloudpi #1097: Debian Buster curl install.sh

Pearmint, 2020-03-04:
Setup: Debian Buster VM/LXT 

Same like #976 but with ncp-previewgenerator missing too.

[Pastebin](https://pastebin.com/t0KV671S) of the terminal.

ovpc, 2020-03-06:
Hi thanks for reporting. 
Things sometimes change upstream, so tried, but could not reproduce this.
Just ran the installer on a fresh Deb10, and it works like a charm.

Got some errors in NC-admin:

> The database is missing some indexes. Due to the fact that adding indexes on big tables could take some time they were not added automatically. By running "occ db:add-missing-indices" those missing indexes could be added manually while the instance keeps running. Once the indexes are added queries to those tables are usually much faster.
>         Missing index "calendarobject_calid_index" in table "oc_calendarobjects_props".
>         Missing index "schedulobj_principuri_index" in table "oc_schedulingobjects".
>     This instance is missing some recommended PHP modules. For improved performance and better compatibility it is highly recommended to install them.
>         imagick

Fixed by running:
> sudo -u www-data /var/www/nextcloud/occ db:add-missing-indices

> sudo apt install php-imagick

> sudo systemctl reload php7.3-fpm.service

ovpc, 2020-03-10:
#976 has been fixed and closed.
Any reason to keep this open?

closed 2020-07-03 by ovpc