nextcloud/nextcloudpi #1102: Backup just in Compressed file Format

elmike75, 2020-03-10:
Fresh install - NC 18.0.1 Debain 10 Amd64 V 1.21.0 

Backup is just possible in compressed Mode also when disable compressed mode. No rights open file after the Backup?

Think its a bug.

ovpc, 2020-03-10:
Thanks for reporting.
Content of ncp.log please (pastebin or similar)
We need to determine: if you or anyone else can reproduce it on another server?
For that, you need to be more specific in steps taken and their result or output, screenshots are helpful too.
Btw, You need sudo to access backups.

elmike75, 2020-03-10:
OK where i find ncp.log? with su i can extract the backup, so thats fine.

qwertyuser1983, 2020-03-13:
> Backup is just possible in compressed Mode also when disabling compressed mode

OK, I'll try it the compressed mode tonight (Indonesia +7).

I don't know if my case is related, but I am currently searching for **why my Nextcloud backup folder empty**, while **the nc-backup reporting successful backup**

 (｡•́︿•̀｡)

Renan

 Environment

- RaspBerry Pi 4 + External HDD 1 TB

- fstab UUID=[my UUID] /media/[username]/MyData     ext4    rw,suid,dev,auto,user,async,exec      0      2, following this tutorial, https://askubuntu.com/questions/678857/fstab-doesnt-mount-with-exec. I chmod owner to $USER

- Ubuntu Server 18.04 + Kubuntu Desktop, taken from https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial, because I need the raspi-config which is missing from the official

- Snap Docker, https://snapcraft.io/docker, which recently "revived from the coma"

-  NextCloudPi Docker, using Docker Compose from  https://ownyourbits.com/2017/11/15/nextcloudpi-dockers-for-x86-and-arm/ and https://docs.nextcloudpi.com/en/how-to-get-started-with-ncp-docker/.

- media/[username]/MyData/NextCloudPi/Data:data

- Manual port forward, SSH, etc feature in NextCloud site working

- All feature in nextcloud.mysite.ext:4443 working except backup, which report success bus the backup folder empty, and nextcloud update, which shall fail and the rollback fail.

closed 2020-03-20 by nachoparker