nextcloud/nextcloudpi #1103: Content Truncated reported (but wrong report ?!)

Larsene, 2020-03-12:
When uploading a file bigger than 2GB, Nextcloud 18.0.1 report a truncated file : 

`{"reqId":"Xmnj6aluPuGJid-6EcaLowAAAQg","level":3,"time":"2020-03-12T07:26:00+00:00","remoteAddr":"172.20.90.113","user":"admin","app":"PHP","method":"MOVE","url":"/remote.php/dav/uploads/admin/web-file-upload-b5ff979fe13830364fd38c75b1415ac5-1583997717039/.file","message":"file_put_contents(): content truncated from 3149629772 to 2147483647 bytes at /var/www/nextcloud/lib/private/Files/Storage/Local.php#503","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.122 Safari/537.36 OPR/67.0.3575.53","version":"18.0.1.3"}
`

But, this is wrong : the file is 2,9Go, totally uploaded, downloaded again, unzipped without any error.  How to be confident in Nextcloud report, when there are real error or wrong report ?

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
 NextCloudPi version  v1.21.0
NextCloudPi image    NextCloudPi_12-20-19
distribution         Raspbian GNU/Linux 10 \n \l
automount            no
USB devices          sda
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      8.6G/917G
rootfs usage         8.6G/917G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    18.0.1.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               11:34

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "11": "***REMOVED SENSITIVE VALUE***"",
            "1": "***REMOVED SENSITIVE VALUE***"",
            "12": ***REMOVED SENSITIVE VALUE***""
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "18.0.1.3",
        "overwrite.cli.url": "https:\/\/nextcloud.SENTIVIE DOMAIN.com\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/var\/www\/nextcloud\/data\/tmp",
        "mail_smtpmode": "sendmail",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "loglevel": "2",
        "log_type": "file",
        "updater.secret": "***REMOVED SENSITIVE VALUE***",
        "maintenance": false,
        "theme": "",
        "app_install_overwrite": [
            "uploaddetails"
        ]
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Thu Mar 12 00:00:08.985057 2020] [mpm_event:notice] [pid 631:tid 3069895184] AH00489: Apache/2.4.38 (Raspbian) OpenSSL/1.1.1d configured -- resuming normal operations
[Thu Mar 12 00:00:08.985082 2020] [core:notice] [pid 631:tid 3069895184] AH00094: Command line: '/usr/sbin/apache2'

```
</details>

<details>
<summary>Database logs</summary>

```

```
</details>

<details>
<summary>Nextcloud logs</summary>

```
{"reqId":"XmnfqKluPuGJid-6EcaHRAAABRQ","level":3,"time":"2020-03-12T07:07:20+00:00","remoteAddr":"172.20.90.113","user":"admin","app":"PHP","method":"GET","url":"/ocs/v2.php/core/whatsnew?format=json","message":"Undefined index: changelogURL at /var/www/nextcloud/core/Controller/WhatsNewController.php#91","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.122 Safari/537.36 OPR/67.0.3575.53","version":"18.0.1.3"}
{"reqId":"Xmnj6aluPuGJid-6EcaLowAAAQg","level":3,"time":"2020-03-12T07:26:00+00:00","remoteAddr":"172.20.90.113","user":"admin","app":"PHP","method":"MOVE","url":"/remote.php/dav/uploads/admin/web-file-upload-b5ff979fe13830364fd38c75b1415ac5-1583997717039/.file","message":"file_put_contents(): content truncated from 3149629772 to 2147483647 bytes at /var/www/nextcloud/lib/private/Files/Storage/Local.php#503","userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/80.0.3987.122 Safari/537.36 OPR/67.0.3575.53","version":"18.0.1.3"}

```
</details>

ovpc, 2020-03-12:
Does not look like it's NCP related. This channel is for NCP only. You should post to the NC channel if its a bug and to forum if need help.

Larsene, 2020-03-12:
> Does not look like it's NCP related. This channel is for NCP only.

With a "real" Nextcloud installation on a small computer, i didn't encounter this error at all. So, i thought this is related to NCP, perhaps linked to the way NCP use NC.

ovpc, 2020-03-12:
I have several NCP installations, none of them have problems uploading big files. More likely a client issue.

nachoparker, 2020-03-20:
thanks for reporting. This is the correct issue to track

https://github.com/nextcloud/server/issues/16431

closed 2020-03-20 by nachoparker