nextcloud/nextcloudpi #1105: New docker installation upgrade failing

comunozalata, 2020-03-15:
Hi, 
I have deplyed ncp container. All started. Then went to ncp-update, all was fine. And when I tried to do ncp-update-nextcloud, I'm getting an error:

`
[ nc-update-nextcloud ]
Current Nextcloud version 16.0.3.0
Available Nextcloud version 18.0.1
Download Nextcloud 18.0.1...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www//nextcloud-bkp_20200315_1584310927.tar.gz generated
Maintenance mode disabled
Install Nextcloud 18.0.1...
Restore apps...
* calendar
* contacts
* files_texteditor
* gallery
* news
* nextcloudpi
* notes
* previewgenerator.orig
* previewgenerator
* tasks
Restore data...
Fix permissions...
Upgrade...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Set log level to debug
Turned on maintenance mode
Exception: Updates between multiple major versions and downgrades are unsupported.
Update failed
Maintenance mode is kept active
Reset log level
Abort
Clean up...
Rolling back to backup /var/www/nextcloud-bkp_20200315_1584310927-16.0.3.0.tar.gz...
Can only restore from ext/btrfs filesystems
Rollback failed! Data left at /var/www/ncp-data.U8p9WY`

please help.
CZ

nachoparker, 2020-03-16:
Duplicate of #1095

closed 2020-03-16 by nachoparker

nachoparker, 2020-03-18:
I pushed NC images with 18.0.2 for all architectures