nextcloud/nextcloudpi #1106: Error during nc-update in function is_more_recent_than after fresh install and nc-restore

BstAA, 2020-03-16:
**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version
distribution         Raspbian GNU/Linux 10 \n \l
automount            no
USB devices          sda sdb
datadir              /media/nc-data/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      28G/222G
rootfs usage         2.4G/59G
swapfile             /var/swap
dbdir                /media/nc-data/ncdatabase
Nextcloud check      ok
Nextcloud version    17.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               16:21

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.10.62",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "9": "cloud.beckstette-aalen.de",
            "20": "192.168.10.62",
            "11": "<!DOCTYPE HTML PUBLIC \"-\/\/IETF\/\/DTD HTML 2.0\/\/EN\">\n<html><head>\n<title>302 Found<\/title>\n<\/head><body>\n<h1>Found<\/h1>\n<p>The document has moved <a href=\"https:\/\/icanhazip.com\/\">here<\/a>.<\/p>\n<\/body><\/html>",
            "21": "cloud.beckstette-aalen.de",
            "22": "pi5.fritz.box",
            "12": "cloud.beckstette-aalen.de"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "17.0.2.1",
        "overwrite.cli.url": "https:\/\/cloud.beckstette-aalen.de\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
[... 114 lines cut ...]
{"reqId":"XVwifU26iMycEiOy0IQ5","level":3,"time":"2020-03-05T01:00:07+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"stat(): stat failed for \/media\/nc-data\/ncdata\/appdata_ocdi5c438i77\/preview\/81789\/384-256.jpg at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#143","userAgent":"--","version":"17.0.2.1"}
{"reqId":"XVwifU26iMycEiOy0IQ5","level":3,"time":"2020-03-05T01:00:07+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"filesize(): stat failed for \/media\/nc-data\/ncdata\/appdata_ocdi5c438i77\/preview\/81789\/384-256.jpg at \/var\/www\/nextcloud\/lib\/private\/LargeFileHelper.php#171","userAgent":"--","version":"17.0.2.1"}
{"reqId":"XVwifU26iMycEiOy0IQ5","level":3,"time":"2020-03-05T01:00:07+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"Undefined index: mtime at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#428","userAgent":"--","version":"17.0.2.1"}
{"reqId":"XVwifU26iMycEiOy0IQ5","level":3,"time":"2020-03-05T01:00:07+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"Undefined index: ino at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#429","userAgent":"--","version":"17.0.2.1"}
{"reqId":"XVwifU26iMycEiOy0IQ5","level":3,"time":"2020-03-05T01:00:07+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"Undefined index: dev at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#430","userAgent":"--","version":"17.0.2.1"}
{"reqId":"Xm5Tfdixk4GkArfVgVvsOQAAUAI","level":2,"time":"2020-03-15T16:10:39+00:00","remoteAddr":"192.168.10.62","user":"--","app":"no app in context","method":"POST","url":"\/index.php\/login","message":"Login failed: klaus (Remote IP: 192.168.10.62)","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:74.0) Gecko\/20100101 Firefox\/74.0","version":"17.0.2.1"}
{"reqId":"Xm5Tvtixk4GkArfVgVvsQgAAVgQ","level":2,"time":"2020-03-15T16:11:42+00:00","remoteAddr":"192.168.10.62","user":"--","app":"no app in context","method":"POST","url":"\/index.php\/login","message":"Login failed: ncp (Remote IP: 192.168.10.62)","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:74.0) Gecko\/20100101 Firefox\/74.0","version":"17.0.2.1"}

```
</details>

**New system setup**

1. Fresh installation 2020-02-13-raspbian-buster-lite
2. Install ncp with bash installer
3. Init and test --> o.k.
4. Move: nc-database ; nc-datadir
5. nc-restore --> nc version is shown as 17.0.2.1 --> o.k.
6. access for data and accounts o.k.
7. nc-update --> error see log

**First Diagnostics**
`/usr/local/etc/ncp-version is missing` ... don't know why

[ncp.log](https://github.com/nextcloud/nextcloudpi/files/4339179/ncp.log)

nachoparker, 2020-03-20:
this has been already fixed. Please try again.

Also you want to `sudo apt-get install -y haveged`

closed 2020-03-20 by nachoparker

BstAA, 2020-03-24:
Sorry for late reply! I had to handle a VM-server breakdown :-(

I couldn't your proposal, as for a quick fix I went a different way to to restore via Raspi-NCP-image.

If of interest, I can take a test run on a development system.
Do you want me to try?

Klaus