nextcloud/nextcloudpi #1107: NextcloudPi armhf docker image broken ?

Seferi, 2020-03-19:
Hi, I installed NextcloudPi on several devices without a problem few days ago. Today it somehow doesn’t work anymore. I see that the image is updated. I wrote in the forums as well and am not the only one. Here is the output of docker logs:

root@DietPi:~# docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /mnt/RaidDrive/ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi-armhf 192.168.178.121
Unable to find image ‘ownyourbits/nextcloudpi-armhf:latest’ locally
latest: Pulling from ownyourbits/nextcloudpi-armhf
fff3167bf8c7: Pull complete
fad9dea8f4af: Pull complete
a76a2cd28d96: Pull complete
ca62447ad296: Pull complete
3d8b05ac9b10: Pull complete
93ad534691af: Pull complete
ecbe463f048c: Pull complete
4f4fb700ef54: Pull complete
b28114bbcf02: Pull complete
6f3b627fdb33: Pull complete
85fe4abd5c7b: Pull complete
753acfc16746: Pull complete
0ed90b45ab62: Pull complete
f6fa8374aaa8: Pull complete
912b7ca15d26: Pull complete
87dc2405471d: Pull complete
acee78231279: Pull complete
8c661ff3eb1e: Pull complete
db0dc6a13e59: Pull complete
9ac0b962e7ff: Pull complete
07326627d55e: Pull complete
dc4900f85f5c: Pull complete
4d9407dbac68: Pull complete
3f85afe31c27: Pull complete
86edf2175f64: Pull complete
0e78debe0572: Pull complete
78ccb61fb967: Pull complete
Digest: sha256:60791a0c9f3e919a8bbc4c3117e542f8f6d7a2d98f3fd3febc2565fee6f47546
Status: Downloaded newer image for ownyourbits/nextcloudpi-armhf:latest
585450269ffa4754ebefb62724d60fea75369e7b9a54911f600ee2e4919fccb4
root@DietPi:~# docker logs -f nextcloudpi
Initializing empty volume…
Making /usr/local/etc/ncp-config.d persistent …
Making /etc/services-enabled.d persistent …
Making /etc/letsencrypt persistent …
Making /etc/shadow persistent …
Making /etc/cron.d persistent …
Making /etc/cron.daily persistent …
Making /etc/cron.hourly persistent …
Making /etc/cron.weekly persistent …
Making /usr/local/bin persistent …
Making /etc/apache2 persistent …
Starting PHP-fpm
Starting Apache
Starting mariaDB
2020-03-18 19:59:37 0 [Note] mysqld (mysqld 10.3.22-MariaDB-0+deb10u1) starting as process 102 …
Starting Redis
Starting Cron
Starting Postfix
Nextcloud is not installed - only a limited number of commands are available

There are no commands defined in the “config:system” namespace.

Init done

unixtan, 2020-03-19:
Having the same problem. Been trying to install for the past 2 days but in vain. It was working fine 3 weeks ago.

Seferi, 2020-03-19:
Latest image (1.23.1) is broken. I just pulled 1.20.1 and it works...

unixtan, 2020-03-19:
Great! Thanks.

nachoparker, 2020-03-20:
thanks for reporting. I was able to reproduce here, looking into it

nachoparker, 2020-03-20:
Uploaded a new image

closed 2020-03-20 by nachoparker

STUKguy, 2020-09-17:
Im assuming the issue has not been fixed with the latest version as i still have the same issue. how do i download 1.20.1