nextcloud/nextcloudpi #1108: dhcpcd again fails to obtain an IPv4 address after unattended reboot

Halloran, 2020-03-20:
[Similar to the previously reported (and fixed) issue #938] 

Running ncp v 1.23.1 on a headless RPi (also keyboardless, which was relevant in the earlier issue). Using only wired ethernet, with a static address being assigned by the dhcp server on the network.

When booting up the pi, dhcpcd hangs for a bit in the boot sequence, though the message indicates that dhcpcd does succeed. Eventually, boot up completes, but the ncp has no IPv4 address. At this point, I can log into the machine from ssh on the LAN via nextcloudpi or nextcloudpi.lan (nextcloudpi.local no longer works, though no idea if that's at all related).

To resolve the issue, I can either log in via ssh and run `sudo dhcpcd`, or  unplug the ethernet cable, reboot, and then plug the cable back in once the boot has completed. But obviously, neither approach works when the reboot needs to happen unattended.

Happy to help out in troubleshooting, etc. as long as I am available to do so (also not got access to a keyboard if that is relevant.

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.23.1
NextCloudPi image    NextCloudPi_09-28-17
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          sda 
datadir              /media/nextCloudDrive/ncdata
data in SD           no
data filesystem      ext2/ext3
data disk usage      939G/3.6T
rootfs usage         5.6G/15G
swapfile             /media/USBdrive/swap
dbdir                /media/USBdrive/ncdatabase
Nextcloud check      ok
Nextcloud version    17.0.2.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               7min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.1.20",
            "3": "hallorancloud.ddns.net",
            "4": "hallorancloud.ddns.net",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/192.168.1.20\/",
        "dbtype": "mysql",
[... 86 lines cut ...]
```

{"reqId":"XnQheU-j-Qv4qW7WCGR7FgAAAAQ","level":4,"time":"2020-03-20T01:50:50+00:00","remoteAddr":"76.15.52.150","user":"maggie","app":"webdav","method":"MKCOL","url":"\/remote.php\/dav\/files\/maggie\/MBA%20Documents\/Microsoft%20User%20Data\/Outlook%20Script%20Menu%20Items\/Sample%20Automator%20Workflows\/Send%20the%20current%20Safari%20Web%20page%20URL%20in%20a%20message%5Cscu.workflow","message":{"Exception":"OCA\\DAV\\Connector\\Sabre\\Exception\\InvalidPath","Message":"File name contains at least one invalid character","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":1178,"function":"getChild","class":"OCA\\DAV\\Connector\\Sabre\\Directory","type":"->","args":["Send the current Safari Web page URL in a message\\scu.workflow"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/CorePlugin.php","line":590,"function":"createCollection","class":"Sabre\\DAV\\Server","type":"->","args":["files\/maggie\/MBA Documents\/Microsoft User Data\/Outlook Script Menu Items\/Sample Automator Workflows\/Send the current Safari Web page URL in a message\\scu.workflow",{"__class__":"Sabre\\DAV\\MkCol"}]},{"function":"httpMkcol","class":"Sabre\\DAV\\CorePlugin","type":"->","args":[{"absoluteUrl":"https:\/\/hallorancloud.ddns.net\/remote.php\/dav\/files\/maggie\/MBA%20Documents\/Microsoft%20User%20Data\/Outlook%20Script%20Menu%20Items\/Sample%20Automator%20Workflows\/Send%20the%20current%20Safari%20Web%20page%20URL%20in%20a%20message%5Cscu.workflow","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/event\/lib\/EventEmitterTrait.php","line":105,"function":"call_user_func_array","args":[[{"__class__":"Sabre\\DAV\\CorePlugin"},"httpMkcol"],[{"absoluteUrl":"https:\/\/hallorancloud.ddns.net\/remote.php\/dav\/files\/maggie\/MBA%20Documents\/Microsoft%20User%20Data\/Outlook%20Script%20Menu%20Items\/Sample%20Automator%20Workflows\/Send%20the%20current%20Safari%20Web%20page%20URL%20in%20a%20message%5Cscu.workflow","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":479,"function":"emit","class":"Sabre\\Event\\EventEmitter","type":"->","args":["method:MKCOL",[{"absoluteUrl":"https:\/\/hallorancloud.ddns.net\/remote.php\/dav\/files\/maggie\/MBA%20Documents\/Microsoft%20User%20Data\/Outlook%20Script%20Menu%20Items\/Sample%20Automator%20Workflows\/Send%20the%20current%20Safari%20Web%20page%20URL%20in%20a%20message%5Cscu.workflow","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":254,"function":"invokeMethod","class":"Sabre\\DAV\\Server","type":"->","args":[{"absoluteUrl":"https:\/\/hallorancloud.ddns.net\/remote.php\/dav\/files\/maggie\/MBA%20Documents\/Microsoft%20User%20Data\/Outlook%20Script%20Menu%20Items\/Sample%20Automator%20Workflows\/Send%20the%20current%20Safari%20Web%20page%20URL%20in%20a%20message%5Cscu.workflow","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Server.php","line":317,"function":"exec","class":"Sabre\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/remote.php","line":163,"args":["\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/apps\/dav\/lib\/Connector\/Sabre\/Directory.php","Line":227,"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Macintosh) mirall\/2.6.0stable (build 20190927) (Nextcloud)","version":"17.0.2.1"}
{"reqId":"XnQheaxu37rYb9W5PHcaDQAAAAU","level":4,"time":"2020-03-20T01:50:50+00:00","remoteAddr":"76.15.52.150","user":"maggie","app":"webdav","method":"MKCOL","url":"\/remote.php\/dav\/files\/maggie\/MBA%20Documents\/Microsoft%20User%20Data\/Outlook%20Script%20Menu%20Items\/Sample%20Automator%20Workflows\/Send%20the%20system%20profile%20in%20a%20message%5Cscp.workflow","message":{"Exception":"OCA\\DAV\\Connector\\Sabre\\Exception\\InvalidPath","Message":"File name contains at least one invalid character","Code":0,"Trace":[{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/Server.php","line":1178,"function":"getChild","class":"OCA\\DAV\\Connector\\Sabre\\Directory","type":"->","args":["Send the system profile in a message\\scp.workflow"]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/dav\/lib\/DAV\/CorePlugin.php","line":590,"function":"createCollection","class":"Sabre\\DAV\\Server","type":"->","args":["files\/maggie\/MBA Documents\/Microsoft User Data\/Outlook Script Menu Items\/Sample Automator Workflows\/Send the system profile in a message\\scp.workflow",{"__class__":"Sabre\\DAV\\MkCol"}]},{"function":"httpMkcol","class":"Sabre\\DAV\\CorePlugin","type":"->","args":[{"absoluteUrl":"https:\/\/hallorancloud.ddns.net\/remote.php\/dav\/files\/maggie\/MBA%20Documents\/Microsoft%20User%20Data\/Outlook%20Script%20Menu%20Items\/Sample%20Automator%20Workflows\/Send%20the%20system%20profile%20in%20a%20message%5Cscp.workflow","__class__":"Sabre\\HTTP\\Request"},{"__class__":"Sabre\\HTTP\\Response"}]},{"file":"\/var\/www\/nextcloud\/3rdparty\/sabre\/event\/lib\/EventEmitterTrait.php","line":105,"function":"call_user_func_array","args":[[{"__class__":"Sabre\\DAV\\CorePlugin"},"httpMkcol"],[{"absoluteUrl":"https:\/\/hallorancloud.ddns.net\/remote.php\/dav\/files\/maggie\/MBA%20Documents\/Microsoft%20User%20Data\/Outlook%20Script%20Menu%20Items\/Sample%20Automator%20Workflows\/Send%20the%20system%20profile%20in%20a%20message%5Cscp.workflow","__class [... cut ...] .php","line":35,"function":"exec","class":"OCA\\DAV\\Server","type":"->","args":[]},{"file":"\/var\/www\/nextcloud\/remote.php","line":163,"args":["\/var\/www\/nextcloud\/apps\/dav\/appinfo\/v2\/remote.php"],"function":"require_once"}],"File":"\/var\/www\/nextcloud\/3rdparty\/doctrine\/dbal\/lib\/Doctrine\/DBAL\/Driver\/PDOStatement.php","Line":117}},"CustomMessage":"--"},"userAgent":"Mozilla\/5.0 (Macintosh) mirall\/2.6.4stable (build 20200303) (Nextcloud)","version":"17.0.2.1"}

```
</details>

Halloran, 2020-03-29:
Behavior is the same in v 1.24.0

Halloran, 2020-04-08:
Also the same behavior in v1.24.2

Jeanoo, 2020-05-30:
Hello

I have the same problem 
 
Have you found a solution?

Halloran, 2020-05-30:
@Jeanoo No. I have tried the solutions that were suggested for #938, but that does not work for this case. So in the meantime, I have been able to manage it with manual intervention.

Jeanoo, 2020-05-30:
Thank you for your response Halloran
Me too, I'm doing the same as you.

Halloran, 2020-06-13:
FWIW, I have upgraded my hardware from a 3 x to a 4 B+ (for reasons unrelated to this issue), and this issue no longer applies to me, so I cannot help with testing at this point.

nachoparker, 2021-10-15:
closing as stale, please reopen if you still see this

closed 2021-10-15 by nachoparker