nextcloud/nextcloudpi #1117: Bug with Cron.php

Pushergene, 2020-03-29:
I am using Docker, 4gb RAM, bad CPU.
When i switch to AJAX, its only saved for ~10min. Probably because it randomly stops working after some minutes, so it switches to Cron.
This is a big problem, especially when u dont have much RAM, then cron does freeze the whole computer.
Expected:
If Nextcloud instances gets frozen caused to cron, then allow us to disable cron background. 
Or:
Fix AJAX and cron, cron gives high cpu

Pushergene, 2020-03-29:
How i fixed it: I just removed cron.php, but it shouldnt be a permanent solution because it gets recreated after each nextcloud update.

hvedemelsbof, 2020-03-29:
Funny how you just posted this, i'm debugging it myself as we speak. In short i've narrowed it down to following using the nextcloud log and mysql log:
- The webinterface writes the new setting to the database, but does NOT overwrite the existing cron-job.
- When the cron-job runs the setting is automatically reset from Ajax to Cron again in the database.

The solution for most people will be to manually remove the cronjob. If i do that it works as intended. The problem gets worse if you're running nextcloud in a docker container (my case). Then the cronjob is rewritten every time the container is restarted. The cronjob is coded in the original docker-file. I tried editing the file -> comitting the container to a new image -> then running that. But the file is still recreated.
I think the problem should actually be addressed to the main nextcloud github - because the cron.php should simply ignore being called, if the setting is Ajax in the database, which is not the case atm. (I'm on nextcloud 17).
Hope that enlightens you - i'm currently looking for the original dockerfile for my armbian distro to recreate the image without the cronjob and fix my problem that way.

closed 2021-01-20 by nachoparker