nextcloud/nextcloudpi #1118: curl installer creates empty /usr/local/bin/ncp

BstAA, 2020-04-04:
**System information**

```
root@cloud:/etc# cat armbian-image-release
# PLEASE DO NOT EDIT THIS FILE
BOARD=rock64
BOARD_NAME="ROCK64"
BOARDFAMILY=rockchip64
VERSION=5.71
LINUXFAMILY=rockchip64
BRANCH=default
ARCH=arm64
IMAGE_TYPE=user-built
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
IMAGE_UUID=f2838a77-f7d0-4566-aa05-927495b604bd
```
```
root@cloud:/etc# cat armbian-release
# PLEASE DO NOT EDIT THIS FILE
BOARD=rock64
BOARD_NAME="ROCK64"
BOARDFAMILY=rockchip64
BUILD_REPOSITORY_URL=https://github.com/armbian/rkbin
BUILD_REPOSITORY_COMMIT=51d55a4
VERSION=5.90
LINUXFAMILY=rockchip64
BRANCH=default
ARCH=arm64
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
```
```
root@cloud:/etc# cat os-release
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
After having lots of problems upgrading from ncp 1.20 to 1.24 (see [https://help.nextcloud.com/t/help-no-data-visible-after-nc-restore/76585](url), decided to run a fresh install with curl installer.

Installation ended without errors but: /usr/local/ is nearly empty:
`root@cloud:/usr/local# tree
```

|-- bin
|   |-- ncc
|   `-- ncp
|-- etc
|   |-- library.sh
|   |-- ncp-config.d
|   |   `-- nc-nextcloud.cfg
|   `-- ncp.cfg
|-- games
|-- include
|-- lib
|   |-- python2.7
|   |   |-- dist-packages
|   |   `-- site-packages
|   `-- python3.7
|       `-- dist-packages
|-- man -> share/man
|-- sbin
|-- share
|   |-- ca-certificates
|   |-- fonts
|   `-- man
`-- src
19 directories, 4 files
```
No ncp command available!

BstAA, 2020-04-04:
After further debugging found

`ERROR: Module authnz_external does not exist!`

and than exiting install script
But

`libapache2-mod-authnz-external is already the newest version (3.3.2-0.1+b1)`

nachoparker, 2021-01-20:
the curl installer expects a minimal clean debian installation. It is not very robust at the moment to handle other situations

closed 2021-01-20 by nachoparker