nextcloud/nextcloudpi #1124: ncp-rsync configure non-root account

mitodeboa, 2020-04-20:
**Description**  
Add the possibility of using non-root account to rsync

**Example**  
Adding the parameter --rsync-path="sudo rsync" to rsync command and configure sudoers in destination server

sunjam, 2020-05-05:
Aren't you able to define a particular user for ncp-rsync besides root?

nachoparker, 2020-05-06:
you select the user by typing it in the first field. See the default suggestion: `user@ip:/path/to/sync`.

closed 2020-05-06 by nachoparker

mitodeboa, 2020-05-06:
Yes, it is possible to define the user but rsync cannot set attributes, acl, time,... even if the user has permissions on the destination directory

mitodeboa, 2020-05-06:
@nachoparker. I have verified that when doing rsync with a non-root user, errors occur when applying the attributes of the file and I have verified that if I add the mentioned parameter, those errors do not occur. This is a constructive contribution is not a request for support or criticism

nachoparker, 2020-05-06:
Yes, thanks for the heads up.

I am pretty sure that by using the user there should be no issues with permissions as long as the rest of the systems are well configured.

We would need to dig a bit deeper into those errors for your particular case, but definitely using `user@` should work since we are simply using `rsync` which uses system users on the remote end.