nextcloud/nextcloudpi #1125: Nextcloud 19 from version 18.0.3

sunjam, 2020-04-21:
### From NextcloudPi beta 18.0.3 on Armbian Odroid C2 to 19.0.0 beta4

Following apps have been disabled: news (incompatible), nextcloudpi (incompatible), previewgenerator (incompatible)

Note: 19.0.0 is running successfully so far.  Obviously, this is a test instance.  Let me know if you want any particular thing checked out.  :+1: 

Preparing update

Set log level to debug

Turned on maintenance mode

Repair step: Repair MySQL collation

Repair info: All tables already have the correct collation -> nothing to do

Repair step: Repair SQLite autoincrement

Repair step: Copy data from accounts table when migrating from ownCloud

Repair step: Drop account terms table when migrating from ownCloud

Updating database schema

Updated database

Repair step: Migrate old user config

[1 / 0]: Migrate old user config

[1 / 1]: Migrate old user config

Updated "accessibility" to 1.5.0

Updated "federatedfilesharing" to 1.9.0

Updated "files_pdfviewer" to 1.8.0

Updated "files_rightclick" to 0.16.0

Updated "files_videoplayer" to 1.8.0

Updated "logreader" to 2.4.0

Updated "password_policy" to 1.9.0

Updated "photos" to 1.1.0

Updated "privacy" to 1.3.0

Updated "provisioning_api" to 1.9.0

Updated "recommendations" to 0.7.0

Updated "serverinfo" to 1.9.0

Updated "settings" to 1.1.0

Updated "support" to 1.2.0

Updated "survey_client" to 1.7.0

Updated "text" to 3.0.1

Repair step: Add background job to check for backup codes

Updated "twofactor_backupcodes" to 1.8.0

Updated "updatenotification" to 1.9.0

[... 318 lines cut ...]
Repair step: Cleanup invalid photocache files for carddav

Repair step: Add background job to cleanup login flow v2 tokens

Repair step: Remove potentially over exposing share links

Repair info: No need to remove link shares.

Repair step: Clear access cache of projects

Repair step: Switches from deprecated "production" to "stable" update channel

Repair step: Sets the enterprise logo

Repair step: Reset generated avatar flag

Starting code integrity check

Finished code integrity check

Turned off maintenance mode

Reset log level

Following apps have been disabled: news (incompatible), nextcloudpi (incompatible), previewgenerator (incompatible)

sunjam, 2020-04-21:
Log from Admin/Administration/Support/System Report:

Server configuration detail

Operating system: Linux 5.4.28-meson64 #20.02.8 SMP PREEMPT Mon Mar 30 09:12:52 CEST 2020 aarch64

Webserver: Apache (fpm-fcgi)

Database: mysql 10.3.22

PHP version: 7.3.11-1~deb10u1

Modules loaded: Core, date, libxml, openssl, pcre, zlib, filter, hash, Reflection, SPL, session, sodium, standard, cgi-fcgi, mysqlnd, PDO, xml, bz2, calendar, ctype, curl, dom, mbstring, fileinfo, ftp, gd, gettext, gmp, iconv, igbinary, intl, json, ldap, exif, mysqli, pdo_mysql, Phar, posix, readline, redis, shmop, SimpleXML, smbclient, sockets, sysvmsg, sysvsem, sysvshm, tokenizer, wddx, xmlreader, xmlwriter, xsl, zip, libsmbclient, Zend OPcache

Nextcloud version: 19.0.0 beta 4 - 19.0.0.5

Updated from an older Nextcloud/ownCloud or fresh install:

Where did you install Nextcloud from: unknown

<details><summary>Signing status</summary>

Array ( )

</details>

<details><summary>List of activated apps</summary>

Enabled:
 - accessibility: 1.5.0
 - activity: 2.12.0
 - bookmarks: 3.0.0-beta.2
 - calendar: 2.0.3
 - cloud_federation_api: 1.2.0
 - comments: 1.9.0
 - contacts: 3.3.0
 - contactsinteraction: 1.0.0
 - dav: 1.15.0
 - federatedfilesharing: 1.9.0
 - federation: 1.9.0
 - files: 1.14.0
 - files_pdfviewer: 1.8.0
 - files_rightclick: 0.16.0
 - files_sharing: 1.11.0
 - files_trashbin: 1.9.0
 - files_versions: 1.12.0
 - files_videoplayer: 1.8.0
 - firstrunwizard: 2.8.0
 - logreader: 2.4.0
 - lookup_server_connector: 1.7.0
 - nextcloud_announcements: 1.8.0
 - notes: 3.2.0
 - notifications: 2.7.0
 - oauth2: 1.7.0
 - password_policy: 1.9.0
 - photos: 1.1.0
 - privacy: 1.3.0
 - provisioning_api: 1.9.0
 - recommendations: 0.7.0
 - serverinfo: 1.9.0
 - settings: 1.1.0
 - sharebymail: 1.9.0
 - support: 1.2.0
 - survey_client: 1.7.0
 - systemtags: 1.9.0
 - tasks: 0.12.1
 - text: 3.0.1
 - theming: 1.10.0
 - twofactor_backupcodes: 1.8.0
 - updatenotification: 1.9.0
 - viewer: 1.3.0
 - workflowengine: 2.1.0
Disabled:
 - admin_audit
 - encryption
 - files_external
 - news
 - nextcloudpi
 - previewgenerator
 - user_ldap

</details>

<details><summary>Configuration (config/config.php)</summary>

{
    "passwordsalt": "***REMOVED SENSITIVE VALUE***",
    "secret": "***REMOVED SENSITIVE VALUE***",
    "trusted_domains": {
        "0": "localhost",
        "11": "52.119.120.83",
        "1": "172.17.0.2",
        "5": "nextcloudpi.local",
        "7": "nextcloudpi",
        "8": "nextcloudpi.lan",
        "20": "192.168.1.101",
        "21": "ncp.local"
    },
    "datadirectory": "***REMOVED SENSITIVE VALUE***",
    "dbtype": "mysql",
    "version": "19.0.0.5",
    "overwrite.cli.url": "http:\/\/localhost",
    "dbname": "***REMOVED SENSITIVE VALUE***",
    "dbhost": "***REMOVED SENSITIVE VALUE***",
    "dbport": "",
    "dbtableprefix": "oc_",
    "mysql.utf8mb4": true,
    "dbuser": "***REMOVED SENSITIVE VALUE***",
    "dbpassword": "***REMOVED SENSITIVE VALUE***",
    "installed": true,
    "instanceid": "***REMOVED SENSITIVE VALUE***",
    "memcache.local": "\\OC\\Memcache\\Redis",
    "memcache.locking": "\\OC\\Memcache\\Redis",
    "redis": {
        "host": "***REMOVED SENSITIVE VALUE***",
        "port": 0,
        "timeout": 0,
        "password": "***REMOVED SENSITIVE VALUE***"
    },
    "tempdirectory": "\/var\/www\/nextcloud\/data\/tmp",
    "mail_smtpmode": "sendmail",
    "mail_smtpauthtype": "LOGIN",
    "mail_from_address": "***REMOVED SENSITIVE VALUE***",
    "mail_domain": "***REMOVED SENSITIVE VALUE***",
    "preview_max_x": "2048",
    "preview_max_y": "2048",
    "jpeg_quality": "60",
    "overwriteprotocol": "https",
    "updater.release.channel": "beta",
    "updater.secret": "***REMOVED SENSITIVE VALUE***",
    "maintenance": false,
    "theme": "",
    "loglevel": 2
}

</details>

External storages: files_external is disabled

Encryption: no

User-backends:

OC\User\Database

Browser: Mozilla/5.0 (X11; Linux aarch64; rv:75.0) Gecko/20100101 Firefox/75.0

ovpc, 2020-08-13:
Hi,
Running nc-update-nextcloud from 18.0,7 to 19.0.1 seems to work fine (tested in a curl installed VM), it just needs/asks for:

>sudo ncc db:add-missing-columns
sudo apt install php-bcmath
sudo apt install php-gmp
sudo apt install php-imagick (at your own risk, devs recommend not to use it)

https://help.nextcloud.com/t/nextcloudpi-and-nextcloud-19/89508/5

sunjam, 2020-08-13:
install php-imagick (at your own risk, devs recommend not to use it)

What did devs say this was a risk?  Do you mean just with deprecated
Gallery app, or with Photo app as well?

ovpc, 2020-08-14:
https://github.com/Imagick/imagick
>We strongly recommend upgrading to the latest version (greater than 6.9.1 or 7.x) if at all possible.

Rasbian/Debian is still on
php-imagick/stable,now 3.4.3-4.1 armhf

zweif, 2021-02-12:
They recommend that ImageMagick version should be greater than 6.9.1. The version in buster is 6.9.10 - so it should be fine, or am I missing something?

ovpc, 2021-02-12:
when i install it in Buster I get:
Setting up php-imagick (3.4.3-4.1)
Setting up libmagickcore-6.q16-6:amd64 (8:6.9.10.23+dfsg-2.1+deb10u1)

zweif, 2021-02-12:
>We strongly recommend upgrading to the latest version (greater than 6.9.1 or 7.x) if at all possible.

I meant this refers to the ImageMagick version which should be 6.9.10 in Buster (8:6.9.10.23+dfsg-2.1+deb10u1), not imagick version (latest version 3.4.4). So I thought it should be fine to include this.

But this refers to possible image corruption issues and not to general security considerations for running ImageMagick in a server environment (https://github.com/Imagick/imagick#security).

Just installing imagick doesn't work in docker (persistance), but I don't know if it's safe to include by default.

closed 2021-05-09 by nachoparker