nextcloud/nextcloudpi #1127: `Can't connect to local MySQL server through socket` when restarting rPi

hunhejj, 2020-04-27:
After restarting the raspberry, NextCloudPi usually has the following error:
`ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (111)`. The server is not available then, but it's docker status is `Up`.

A `docker restart <nextcloudpi_container>` resolves the issue. Then it starts as expected. 

**System information**
Raspberry Pi4
NextCloudPi version: `v1.25.0`
Docker image: `NextCloudPi_docker_03-20-20` 

The volume is mounted (btw. data is stored on a external disk):
```
volumes:
      - /mnt/hdd1/ncData:/data
      - /etc/localtime:/etc/localtime:ro
```

### 2 questions
1. Any ideas why this could happen?
2. Shouldn't the container exit if it can't connect to the DB? Then my restart policy could handle the issue.

ovpc, 2020-04-27:
Looks like db is on external too and not mounted yet (fstab) when docker starts. Should use forum tho, Github is for development, discussing bugs and documentation only.

hunhejj, 2020-04-27:
Because of the mounted volume (see above) the db is probably on the external drive too, yes.

hunhejj, 2020-04-27:
Isn't it a bug that the container seems to be up although the DB was not started?

ovpc, 2020-06-19:
Closing this, no-one has reproduced this afaik.

Coernel82, 2020-07-14:
@hunhejj Is it a Rapsberry Pi 4 with **8 GB**? I get the suspicion that the
`ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2)`
ineed is a bug and that this thread needs to be reopened. [This source](https://forum-raspberrypi.de/forum/thread/47927-rpi-4-8gb-inkompatibel-mit-nextcloud/) also indicates it should be a real problem - but for the 8 GB version only.

hunhejj, 2020-07-14:
I had the issue with a raspberry pi 4 with 4GB

ovpc, 2020-07-14:
Reopening an issue does not make it a bug.
Anyone else reproducing this ?  on rpi4-8Gb, its a docker container too, which add's another layer

markgllin, 2020-07-14:
I'm currently running into the same issue on RPi4-2GB in the docker container.

From `docker logs`:
```
2020-07-14 23:01:32 0 [Note] mysqld (mysqld 10.3.22-MariaDB-0+deb10u1) starting as process 106 ...
Starting Redis
Starting Cron
Starting Postfix
ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (111)
Init done
```

my `docker-compose.yml`:
```
services:
  nextcloud:
    image: ownyourbits/nextcloudpi-armhf
    container_name: nextcloud
    volumes:
      - ncdata:/data
      - /etc/localtime:/etc/localtime:ro
    restart: unless-stopped
    security_opt:
      - no-new-privileges:true
    ports:
      - 81:80
      - 445:443
      - 4443:4443

volumes:
   ncdata:
     driver: local
     driver_opts:
       type: 'none'
       o: 'bind'
       device: '/mnt/docker/docker/data'
```
what's curious is that it ran successfully the first time i did a `docker-compose up -d`, correctly creating the bindmount where the db is stored. 

`ncp-report` didn't seem to capture anything 
```
root@11eed60a5e62:/# ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

will continue debugging and report back if i find anything

Coernel82, 2020-07-16:
OK, I do my best for a bug report:

# Steps to reproduce
After multiple tries using the NextcloudPi image without being able to connect via SSH (I have no micro-hdmi cable to check whats happening on the device) I decided to try the Debian-Script:

<details>
<summary>Installing image, firstboot and Nextcloudpi-Install-Script</summary>

1. Downloading Raspian OS Image and verify checksum 
2. Writing Raspberry Pi OS (32-bit) Lite 2020-05-27 to SanDisk Ultra 32GB HC I Class 10 SD-Card and verify using Etcher.
3. copy ssh and wpa_supplicant.conf to boot partition for ssh and wifi access
4. boot up Raspberry Pi 4B (8 GB version)
5. log in by SSH
6. ```sudo apt-get update && sudo apt-get upgrade```
7. ```curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash``` Note: **sudo** bash!
</details>

<details>
<summary>Trying to set up NextcloudPi</summary>

1. Open 192.168.178.29 without Port in Chrome, ignoring SSL warning
2. click acitvate
3. Error message loop: ```NextCloudPi not yet initialized, trying again in a few secons...```
4. via ssh ```sudo ncp-config'''
5. start NextcloudPi update by clicking Yes
</details>

# Output from ncp-report
```sudo ncp-report```

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.18.0
distribution         Raspbian GNU/Linux 10 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      2.3G/29G
rootfs usage         2.3G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      down
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            wlan0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               1:37

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

[... 96 lines cut ...]
Max resident set          unlimited            unlimited            bytes
Max processes             61664                61664                processes
Max open files            16364                16364                files
Max locked memory         65536                65536                bytes
Max address space         unlimited            unlimited            bytes
Max file locks            unlimited            unlimited            locks
Max pending signals       61664                61664                signals
Max msgqueue size         819200               819200               bytes
Max nice priority         0                    0
Max realtime priority     0                    0
Max realtime timeout      unlimited            unlimited            us
Core pattern: core

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"XOVkQGCtQysnxW4XhIzA","level":3,"time":"2020-07-16T13:15:09+00:00","remoteAddr":"","user":"--","app":"cron","method":"","url":"--","message":{"Exception":"Exception","Message":"Not installed","Code":0,"Trace":[{"file":"/var/www/nextcloud/lib/base.php","line":651,"function":"checkInstalled","class":"OC","type":"::","args":[]},{"file":"/var/www/nextcloud/lib/base.php","line":1089,"function":"init","class":"OC","type":"::","args":[]},{"file":"/var/www/nextcloud/cron.php","line":42,"args":["/var/www/nextcloud/lib/base.php"],"function":"require_once"}],"File":"/var/www/nextcloud/lib/base.php","Line":282,"CustomMessage":"--"},"userAgent":"--","version":""}

```
</details>

Coernel82, 2020-08-13:
A user in the German Raspberry-Pi-Forum gave a hint which might solve the problem:
https://forum-raspberrypi.de/forum/thread/47927-rpi-4-8gb-inkompatibel-mit-nextcloudpi/?postID=444544#post444544

Basically what he describes is that there is to much memory being allocated thus he decreases the allocated memory.

Original **English** thread from the nextcloud forums: https://help.nextcloud.com/t/cannot-allocate-memory-for-bufferpool-mariadb-on-fresh-raspberry-pi-4-8gb/87067/6

The mentionied file `nc-limits.sh` can be found in `/usr/local/bin/ncp/CONFIG`

nachoparker, 2020-08-13:
looks similar to what we already fixed here

https://github.com/nextcloud/nextcloudpi/issues/1141

nachoparker, 2020-08-30:
can't reproduce here with the latest container

closed 2020-08-30 by nachoparker

crooowned, 2020-12-10:
Having the same issue with the 8gb Version. Force recovery not working, tried reflashing multiple times. Happens after rebooting multiple times. "sudo reboot now" (Not force reboots by unplugging)