nextcloud/nextcloudpi #1129: Nextcloud output after upgrade to beta 19.0.0 beta5

sunjam, 2020-04-30:
NextcloudPi on Odroid C2 test instance.

There are some warnings regarding your setup.

    The database is missing some optional columns. Due to the fact that adding columns on big tables could take some time they were not added automatically when they can be optional. By running "occ db:add-missing-columns" those missing columns could be added manually while the instance keeps running. Once the columns are added some features might improve responsiveness or usability.
        Missing optional column "reference_id" in table "oc_comments".
    This instance is missing some recommended PHP modules. For improved performance and better compatibility it is highly recommended to install them.
        bcmath
        imagick

Co0p3r21, 2020-05-02:
You can just do what is said in the warning execute "occ db:add-missing-columns" and "use forums please this is not for support".

nachoparker, 2020-05-03:
thanks for the heads up and testing @sunjam, I'll add this command to the upgrade process

nachoparker, 2020-05-03:
@Co0p3r21 many thanks for the help. For most users what you did is the correct thing and it helps, I appreciate it. In this case though, @sunjam is an active tester and contributor, so I appreciate his input :)

nachoparker, 2020-05-03:
btw I wonder what app requires `bcmath`... I'll add it if needed when I test this by myself

Co0p3r21, 2020-05-03:
Sorry if I was a little bit over the top. I didn't mean to.

nachoparker, 2020-05-03:
no, actually that's great, we need help moderating github and the forums :)

thanks

cscholz, 2020-05-16:
I've updated from NC 18 to NC19 RC2 and I get the same message:

> Fehlende optionale Spalte "reference_id" in der Tabelle "oc_comments".

occ did not fixed it:

```
sudo -u  $user ./occ db:add-missing-indices
Check indices of the share table.
Check indices of the filecache table.
Check indices of the twofactor_providers table.
Check indices of the login_flow_v2 table.
Check indices of the whats_new table.
Check indices of the cards table.
Check indices of the cards_properties table.
Check indices of the calendarobjects_props table.
Check indices of the schedulingobjects table.
Check indices of the oc_properties table.
Done.
```
PHP and browser cache already cleared.
Is there anything else I can do?

OSevangelist, 2020-06-05:
Same over here. After updating from 18.0.4 to 19.0.0 i executed 

`./occ db:add-missing-indices
`
and cleared the cache but the error remains the same.

sunjam, 2020-06-06:
> Same over here. After updating from 18.0.4 to 19.0.0 i executed
> 
> `./occ db:add-missing-indices `
> 
> and cleared the cache but the error remains the same.

Try restarting Nextcloud.

DeCorrino, 2020-08-28:
Hi,

this warning says: Missing Column
It does NOT say: Missing Index

So yo have to execute: ./occ db:add-missing-columns

Kind regards
Mario

nachoparker, 2020-08-29:
pushed NC19.0.1 for testing, your comments were very useful, thanks.

It can be tested by running `ncp-update devel` and then either waiting for autoupgrade if it's enabled, or run `nc-update-nextcloud` with version `0`.

ovpc, 2020-08-29:
Tried updating on my test pi but got this (I think unrelated to v1.29.0) error, 
Running nc-update-nextcloud
Nextcloud backup directory found. Interrupted installation?
Done. Press any key...

So used NC's native updater to update to the proposed 19.0.2 

Appart from having to run 
> sudo ncc db:add-missing-indices
> sudo ncc db:add-missing-columns

All seems fine, only Two Factor e-mail provider app gets disabled for being untested.

nachoparker, 2020-08-29:
great, thanks oliver. The ncc commands are part of the upgrade so we should be good

nachoparker, 2020-08-29:
published. Closes #1129

ovpc, 2020-08-30:
Running nc-update-nextcloud
Current   Nextcloud version 18.0.7.1
Available Nextcloud version 18.0.8
Download Nextcloud 18.0.8...
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www//nextcloud-bkp_20200830_1598769774.tar.gz generated
Maintenance mode disabled
Install Nextcloud 18.0.8...
Restore apps...
 * appointments
 * calendar
 * contacts
 * deck
 * files_texteditor
 * gallery
 * groupfolders
 * guests
 * mail
 * news
 * nextcloudpi
 * notes
 * ojsxc
 * onlyoffice
 * previewgenerator
 * previewgenerator.orig
 * sharepoint
 * spreed
 * tasks
 * timetracker
 * twofactor_admin
 * twofactor_email
 * twofactor_gateway
 * twofactor_yubikey
Fix permissions...
Upgrade...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available
You may use your browser or the occ upgrade command to do the upgrade
Set log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Checking for update of app accessibility in appstore
Checked for update of app "accessibility" in appstore 
Checking for update of app activity in appstore
Checked for update of app "activity" in appstore 
Checking for update of app calendar in appstore
Checked for update of app "calendar" in appstore 
Checking for update of app cloud_federation_api in appstore
Checked for update of app "cloud_federation_api" in appstore 
Checking for update of app comments in appstore
Checked for update of app "comments" in appstore 
Checking for update of app contacts in appstore
Checked for update of app "contacts" in appstore 
Checking for update of app dav in appstore
Checked for update of app "dav" in appstore 
Checking for update of app federatedfilesharing in appstore
Checked for update of app "federatedfilesharing" in appstore 
Checking for update of app federation in appstore
Checked for update of app "federation" in appstore 
Checking for update of app files in appstore
Checked for update of app "files" in appstore 
Checking for update of app files_external in appstore
Checked for update of app "files_external" in appstore 
Checking for update of app files_pdfviewer in appstore
Checked for update of app "files_pdfviewer" in appstore 
Checking for update of app files_rightclick in appstore
[... 87 lines cut ...]
Check indices of the login_flow_v2 table.
Check indices of the whats_new table.
Check indices of the cards table.
Check indices of the cards_properties table.
Check indices of the calendarobjects_props table.
Check indices of the schedulingobjects table.
Done.

                                                    
  Command "db:add-missing-columns" is not defined.  
                                                    
  Did you mean one of these?                        
      db:add-missing-indices                        
      db:convert-filecache-bigint                   
      db:convert-mysql-charset                      
      db:convert-type                               
                                                    

Abort
Clean up...
Rolling back to backup /var/www/nextcloud-bkp_20200830_1598769774-18.0.7.1.tar.gz...
check free space...
extracting backup file /var/www/nextcloud-bkp_20200830_1598769774-18.0.7.1.tar.gz...
restore files...
restore database...

ovpc, 2020-08-30:
I think I know what went wrong
ncp v1.29.0 is for updateing to 19.0.2 not 18.0.8

closed 2020-08-30 by ovpc

nachoparker, 2020-08-30:
I see, will fix. Thanks

jcb-tn, 2020-09-04:
Problem not solved. 
Updated to Nextcloud 19.02 via stable channel. but get the error furthermore.

[image]

nachoparker, 2020-09-05:
you need to update through ncp-web