nextcloud/nextcloudpi #1130: NextCloudPi’s default settings slow down Talk App

HenryDunham, 2020-05-03:
When using Talk on NCP, everything is getting very slow as soon as two participants join the same conversation. Every single action takes approx. 20s. This is caused by NCP default setting of PHP threads =4. Setting it to 6 or higher solves the issue. (At least for two concurrent users)

For more details, see https://help.nextcloud.com/t/nextcloudpi-and-talk-configuration-incompatible/79469

I suggest that the nc-limits “PHP threads” default setting is chosen based on available RAM size rather than number of CPU cores. (As described here: https://www.kinamo.be/en/support/faq/determining-the-correct-number-of-child-processes-for-php-fpm-on-nginx)

I’ve set the limit for my system to 25. I consider this a very conservative value for NCP on a Raspberry Pi 4B with 4GB. But it’s probably too much for older models…

Steps to reproduce:
- Install NCP based on “NextCloudPi_RPi_03-28-20.tar.bz2”
- Activate Talk App (coTURN not required if you test in LAN)
- Create a new conversation in Talk and share the link
- Join the conversation simultaneously from two independent devices (e.g. your PC and your mobile) -> try to send messages, change names or perform other actions. Everything is very slow and takes up to 20s.
- Set PHP threads to 6 or higher -> restart and test again-> no delay anymore

btw @nachoparker: Thank you for your hard work and NextCloudPi! It’s easy to set up, it’s working great and makes my life much easier. :thumbsup:

nachoparker, 2020-05-03:
hi, 

thanks for sharing. What does your php config file look like?

HenryDunham, 2020-05-04:
Short answer: Yes, the only change was PHP threads = 6

More details: I had to change and test a lot of things to find the root cause. But once I was sure that PHP threads limit causes the problem, I started from scratch with a clean installation of “NextCloudPi_RPi_03-28-20.tar.bz2”. Activating Talk and setting PHP threads to 6 were the only changes I made to the default NCP setup.

But just to be sure, I’ve attached my configuration files from \etc\php\7.3\fpm:
[php_ini.txt](https://github.com/nextcloud/nextcloudpi/files/4575593/php_ini.txt)
[php-fpm_conf.txt](https://github.com/nextcloud/nextcloudpi/files/4575596/php-fpm_conf.txt)
[www_conf.txt](https://github.com/nextcloud/nextcloudpi/files/4575597/www_conf.txt)

nachoparker, 2020-05-04:
great, that can be added easily. I've also been toying with the idea of packaging coturn with NCP, maybe Talk as well out of the box

nachoparker, 2020-08-30:
I set the minimum to 6 for now, thanks!

nachoparker, 2020-08-30:
related https://github.com/nextcloud/nextcloudpi/issues/1068

closed 2020-08-30 by nachoparker