nextcloud/nextcloudpi #1131: _ character included in password field for nc-admin and nc-passwd

sunjam, 2020-05-04:
Confusing from `ncp-config` menu because `_` is actually there and appended to whatever the admin types.  If it is not deleted, it will be treated as part of either password when reset from this menu, which is confusing.
[image]

This menu item also claims to allow one to view the current password, but that is not possible afaik.

ovpc, 2020-05-05:
Hi Sunjam
That is what it does for me
I can view and change the password.
Not sure how yours can only show "_" The first time nc-init runs, passwords for nc-admin (NC) and nc-passwd (NCP-web:4443) are generated.
[image]

nachoparker, 2020-05-05:
Yes, that's not an ideal solution, I kind of hacked it together real fast, I'll try to look again for a better way, but don't have much time these days..

Thanks

nachoparker, 2020-05-05:
> This menu item also claims to allow one to view the current password, but that is not possible afaik.

we don't store passwords anymore for security purposes

sunjam, 2020-05-05:
The only issue is the password fields have "_" in them when editing from `ncp-config`.  If the password fields were empty there would be no issue. ❤️

nachoparker, 2020-05-06:
yeah that was introduced because empty values dont work with the TUI library as a quick cheap bad workaround

sunjam, 2020-05-06:
> yeah that was introduced because empty values dont work with the TUI library as a quick cheap bad workaround

Cool, in that case I'd just recommend `REPLACE`