nextcloud/nextcloudpi #1132: Nextcloudpi - errors with nc-datadir

ncpfd, 2020-05-05:
**Original Problem**
following situation: I’m running NextCloudPi v1.25.0 on a Raspberry Pi 4. Data is on an external HDD. The server worked perfectly until I decide to move datadir to a SDD.
So I shutdown the R-Pi, unplug the HDD, plug in SDD and reboot. Then format SDD via web interface with BTRFS. Great. Of course always applying proper labels as per wiki.
Then I shutdown, plug in both HDD and SDD. Both are mounted and after reading some forum thread execute nc-datadir on the SDD, expecting that datadir is copied to SDD.

However, I get error messages and it doesn’t work. So I decide to shutdown, unplug SDD and reboot again. BUT now NCP doesn’t mount my orginal HDD anymore.
How do I fix this? Tried mounting via command-line but there is no mount target.
NCP webinterface still shows datadir with the original HDD folder, which what it should be, but the drive is not mounted

**nc-datadir Problem**
After this I did a complete re-install from scratch, then re-installed backup without data.
Now the original HDD cannot be mounted (see above), but I also cannot move datadir to either SD-card or an empty external drive (which is mounted just fine).

Output I get:
[ nc-datadir ]
PHP Warning: fileperms(): stat failed for /media/if-ncp-hdd1/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 83
{"reqId":"AHSEIV85U8SxK0O5RKBW","level":3,"time":"2020-05-05T06:47:42+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for /media/if-ncp-hdd1/ncdata/nextcloud.log at /var/www/nextcloud/lib/private/Log/File.php#83","userAgent":"--","version":"18.0.3.0"}
{"reqId":"AHSEIV85U8SxK0O5RKBW","level":2,"time":"2020-05-05T06:47:42+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /media/if-ncp-hdd1/ncdata/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
PHP Warning: fileperms(): stat failed for /media/if-ncp-hdd1/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 83
{"reqId":"AHSEIV85U8SxK0O5RKBW","level":3,"time":"2020-05-05T06:47:42+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for /media/if-ncp-hdd1/ncdata/nextcloud.log at /var/www/nextcloud/lib/private/Log/File.php#83","userAgent":"--","version":"18.0.3.0"}
{"reqId":"AHSEIV85U8SxK0O5RKBW","level":2,"time":"2020-05-05T06:47:42+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
Nextcloud is in maintenance mode - no apps have been loaded

PHP Warning: fileperms(): stat failed for /media/if-ncp-hdd1/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 83
{"reqId":"AHSEIV85U8SxK0O5RKBW","level":3,"time":"2020-05-05T06:47:42+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for /media/if-ncp-hdd1/ncdata/nextcloud.log at /var/www/nextcloud/lib/private/Log/File.php#83","userAgent":"--","version":"18.0.3.0"}
{"reqId":"AHSEIV85U8SxK0O5RKBW","level":3,"time":"2020-05-05T06:47:42+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"chmod(): No such file or directory at /var/www/nextcloud/lib/private/Log/File.php#84","userAgent":"--","version":"18.0.3.0"}
Error reading data directory. Is NextCloud running and configured?

**System information**
<details>
<summary>NextCloudPi diagnostics</summary>

```

PHP Warning:  fileperms(): stat failed for /media/if-ncp-hdd1/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 83
{"reqId":"0QIaXdnA8ksDxVTCSXHQ","level":3,"time":"2020-05-05T06:52:10+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for /media/if-ncp-hdd1/ncdata/nextcloud.log at /var/www/nextcloud/lib/private/Log/File.php#83","userAgent":"--","version":"18.0.3.0"}
{"reqId":"0QIaXdnA8ksDxVTCSXHQ","level":2,"time":"2020-05-05T06:52:10+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /media/if-ncp-hdd1/ncdata/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
PHP Warning:  fileperms(): stat failed for /media/if-ncp-hdd1/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 83
{"reqId":"0QIaXdnA8ksDxVTCSXHQ","level":3,"time":"2020-05-05T06:52:10+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for /media/if-ncp-hdd1/ncdata/nextcloud.log at /var/www/nextcloud/lib/private/Log/File.php#83","userAgent":"--","version":"18.0.3.0"}
{"reqId":"0QIaXdnA8ksDxVTCSXHQ","level":2,"time":"2020-05-05T06:52:10+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"","url":"--","message":"Temporary directory /var/www/nextcloud/data/tmp is not present or writable","userAgent":"--","version":"18.0.3.0"}
Nextcloud is in maintenance mode - no apps have been loaded

PHP Warning:  fileperms(): stat failed for /media/if-ncp-hdd1/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 83
{"reqId":"0QIaXdnA8ksDxVTCSXHQ","level":3,"time":"2020-05-05T06:52:10+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"fileperms(): stat failed for /media/if-ncp-hdd1/ncdata/nextcloud.log at /var/www/nextcloud/lib/private/Log/File.php#83","userAgent":"--","version":"18.0.3.0"}
{"reqId":"0QIaXdnA8ksDxVTCSXHQ","level":3,"time":"2020-05-05T06:52:10+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"chmod(): No such file or directory at /var/www/nextcloud/lib/private/Log/File.php#84","userAgent":"--","version":"18.0.3.0"}
NextCloudPi version  v1.25.0
NextCloudPi image    NextCloudPi_03-28-20
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          none
datadir              /media/if-ncp-hdd1/ncdata (doesn't exist)
rootfs usage         2.4G/7.1G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       no
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               16:36

[... 55 lines cut ...]
[Tue May 05 07:41:41.017062 2020] [authz_host:error] [pid 31694:tid 2830918688] [client 192.168.169.122:55133] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Tue May 05 07:41:41.017063 2020] [authz_host:error] [pid 31694:tid 2839311392] [client 192.168.169.122:55133] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Tue May 05 07:41:43.229435 2020] [authz_host:error] [pid 31694:tid 2805732384] [client 192.168.169.122:55133] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name
[Tue May 05 07:41:44.198734 2020] [authz_host:error] [pid 31694:tid 2805732384] [client 192.168.169.122:55133] AH01753: access check of 'localhost' to /ncp-launcher.php failed, reason: unable to get the remote host name

```
</details>

<details>
<summary>Database logs</summary>

```

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

tail: cannot open '/media/if-ncp-hdd1/ncdata/nextcloud.log' for reading: No such file or directory

```
</details>

ovpc, 2020-05-05:
nc-datadir worked fine last time I used it, but that is some time ago.
Also I used it to move the data from its original location, inside /var/www/nextcloud/data to an external drive mounted at /media/MyLabel/
Not sure if it can and is meant be used to move data from one external to another. I would expect it to, so looks like you ran into permission issue. I usually create a folder ncdata at the new mointpoint and give ownership to www-data. before rumnning nc-datadir. Sometimes unplugging and replugging without proper unmounting fs, can damage data, en mount readonly after a reboot.

Can also try to use rsync to copy the data manually, then edit config to point to it and run nc-scan .

If this can not be reproduced, it is likely not a bug,
You should then, please take it to forum if want further assist, and close this thread

ncpfd, 2020-05-05:
Hi Oliver, 
I suspect there is a bug with nc-datadir, but I'm not sure that's why I'm posting it here. 
It already spent some time searching the forums and understand that nc-datadir will copy the files to the new datadir location. According to others, this should also work from external drive to external drive.
However, that's not my only problem. I can't turn it back to the standard /var/www/nextcloud/data on the SD-card either. So I didn't use any other tools than NCP webinterface, so this shouldn't go wrong.
I've always properly shutdown or unmounted as far as I can tell. 

Not sure how to help replicate this however. My understanding of Nextcloud under the hood is limited.

nachoparker, 2020-05-06:
We would have to see the errors in the original output, the first time nc-datadir is run. AFAIK there are no other reports of nc-datadir stopping working and there have been no recent changes.

ncpfd, 2020-05-06:
How can I generate that output?

If you feel it's better to continue the discussion in the forum, can you recommend a link / subforum to post it?

ovpc, 2020-05-06:
Use
https://help.nextcloud.com/c/support/appliances-docker-snappy-vm
in addition add the "ncp" tag
might find more info in ncp.log and by running ncp-report

ncpfd, 2020-05-08:
I posted here:
https://help.nextcloud.com/t/nextcloudpi-errors-with-nc-datadir-usb-drive-not-mounting-anymore/79851

Could this be some python issue?

ovpc, 2020-06-19:
Is in progress on forum post, closing here for now.

closed 2020-06-19 by ovpc