nextcloud/nextcloudpi #1136: Initializing NextCloudPi for the first time error

cryptopher17, 2020-05-11:
I have installed your amazing docker many times. 
This time round on a brand new server I'm not getting past this message. I can browse the Pi Panel but not NC. 
Attached is part of the log. Please can you advise what I can try to get it back up as I really depend on your container as it has everything I need :) Thanks for the hard work.

nextcloudpi enabled
Error: Could not download app calendar
Init done
stopping Cron...
stopping Redis...
stopping Postfix...
stopping logs...
tail: no process found
Stopping apache
Stopping PHP-fpm
Stopping mariaDB
LAMP cleanup complete
Starting PHP-fpm
Starting Apache
Starting mariaDB
2020-05-11 16:00:24 0 [Note] mysqld (mysqld 10.3.22-MariaDB-0+deb10u1) starting as process 222 ...
Starting Redis
Starting Cron
Starting Postfix
System config value trusted_domains => 1 set to string 172.17.0.2
System config value trusted_domains => 6 set to string 192.168.1.3
Provisioning
Provisioning Redis password
Updating NextCloud config with Redis password
Provisioning MariaDB password
Updating NextCloud config with MariaDB password
Running nc-limits
Using 12520992768B for PHP
Using 4 PHP threads
Using 6677862809 memory for the database
Stopping MariaDB database server: mysqld.
Starting MariaDB database server: mysqld.
Stopping redis-server: redis-server.
Starting redis-server: redis-server.
Init done

ovpc, 2020-05-11:
Error: Could not download app calendar
App server maybe down or time-ing out, just wait 5 min and try again.
Use [forum](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) for assistance, Github is only for reproducible bugs, discussing docs or new features.

nachoparker, 2020-12-02:
The latest images come pre-initialized so this won't happen to anybody again

wsegatto, 2021-02-24:
Nope, still happening.

codeclinic, 2021-05-22:
Urm... It's happening on my new install on a new mac using docker.

Jaeminz, 2021-07-01:
Same here. Stucking on Initilization with latest docker version and datadirecty on external drive.

wsegatto, 2021-07-01:
I was able to solve this issue for me by updating to the latest v18 release (Docker comes with this version, but it's unsupported), and then updated it to the latest v19 release. 
Then it worked.

generalxhd, 2021-07-16:
@wsegatto 's solution worked for me. Go to "nc-update-nextcloud" in the web gui, enter 19.0.0 and then click apply.

kant-chen, 2021-08-07:
> @wsegatto 's solution worked for me. Go to "nc-update-nextcloud" in the web gui, enter 19.0.0 and then click apply.

This works for me as well.

theCalcaholic, 2021-08-07:
I reopened, since this definitely sounds like a big we need to fix. I'm glad that you found a workaround, though

nachoparker, 2021-08-07:
again, the latest image on dockerhub comes pre-initialized. I have no idea what you are all downloading.

closed 2021-08-07 by nachoparker

kant-chen, 2021-08-08:
> again, the latest image on dockerhub comes pre-initialized. I have no idea what you are all downloading.

This should be the correct command to pull the latest image, right?
`sudo docker run -d -p 4443:4443 -p 80:80 -p 443:443 -v ncdata:/data --name nextcloudpi ownyourbits/nextcloudpi <domain name>`
So didn't you face this issue when you do a fresh install?
Is the issue happen randomly on random people?

theCalcaholic, 2021-08-08:
@kant-chen I at least can't reproduce it. Did you execute `docker pull ownyourbits/nextcloudpi`?

kant-chen, 2021-08-09:
> @kant-chen I at least can't reproduce it. Did you try to execute `docker pull ownyourbits/nextcloudpi` (or add `--pull always` to the docker run command)?

I didn't use `docker pull` command or `--pull always` option, as I downgraded to 19.0.0 and already uploaded my backups, I would probably stay on this version for now.
By the way, I think it might be helpful to see the docker image details.
Here is the details of my image:
```
pi@k3s-master:~ $ docker images
REPOSITORY                TAG        IMAGE ID       CREATED         SIZE
ownyourbits/nextcloudpi   latest     4698f1b6dba9   6 months ago    1.05GB
```

```
pi@k3s-master:~ $ docker image inspect 4698f1b6dba9

[
    {
        "Id": "sha256:4698f1b6dba92c47832646c6756e37714e698236c4d253fc63b5ebc5480c4dfc",
        "RepoTags": [
            "ownyourbits/nextcloudpi:latest"
        ],
        "RepoDigests": [
            "ownyourbits/nextcloudpi@sha256:498cd4ce7926e8e018d63e7e60d75bbf536ab1f7ddbbb7094a006c0b13f7bf15"
        ],
        "Parent": "",
        "Comment": "buildkit.dockerfile.v0",
        "Created": "2021-01-18T22:55:40.194168043-07:00",
        "Container": "",
        "ContainerConfig": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": null,
            "Cmd": null,
            "Image": "",
            "Volumes": null,
            "WorkingDir": "",
            "Entrypoint": null,
            "OnBuild": null,
            "Labels": null
        },
        "DockerVersion": "",
        "Author": "",
        "Config": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "ExposedPorts": {
                "443/tcp": {},
                "80/tcp": {}
            },
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": [
                "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin",
                "DOCKERBUILD=1"
            ],
            "Cmd": null,
            "ArgsEscaped": true,
            "Image": "",
            "Volumes": null,
[... 38 lines cut ...]
                "sha256:7695d520fc002503ab352b92153bfe7361d44d0e672b8434118e37da6ffc134f",
                "sha256:a4d606d60d02256635f8dfd0a7c1611daa26d78da58cca862a099a4e02889a01",
                "sha256:41ea216be1ceff3b8561166a8d21fa15c807bd3d393e09c575ed299a12c8bed9",
                "sha256:a05aa9c896e87ce914c099bd00bac194f73ddb1ecd7e5453fb81b33a426a735b",
                "sha256:17c4b979126689426113556a75e9bd1f411a019c0ac362db7a77b1e70e744cef",
                "sha256:97d63e4843d2fbe9b40e65647a6d39a4078b0d201aa3079f32a9de4b8ae12fce",
                "sha256:3513b38c41025b7d6496aa60f5a8c99410ceda85a7c158ab363ec1ab7607cebd",
                "sha256:d49cf1a7c85df9973f6ba4c58ad1c27bb31918b120b53b70a1888990c60ccfd6",
                "sha256:14253e6dd7b92b40c49432bd9bda7109f77b22190ab07c08cacd7a0849b14a2c",
                "sha256:eb36e1334fbc8ae10a21413e7adf9db893cd77c4582a48d758f9fe24505a2f25",
                "sha256:e782be5cc7908127c3c73fbb83dd4bbb388970d7aa481455c1216295e78b77ba",
                "sha256:7b2b8ee597fa9cd75684a02a3b22c4c8ad86de98a6e218c843f0bb3ebb9fc1d1",
                "sha256:d158533e92eecc6b08206b73fc0859b20e8a3565b72fc3e2f83a1816342ac59e",
                "sha256:31295ad77d25b3186d2f89fe1daa1637668917182d8cf51672800d508c9423be",
                "sha256:3753fce70edde60b94b032fd339602493364d34b6c6cff23230bb26b0cceec57",
                "sha256:d7e2a4d3d73d54d727aad66d22d11fce34e9a03340dbba86367f8791d8608d40",
                "sha256:b141712125a2cd9a75a17f42d82c7186a8fa56ed9e63adb560176ffdc41131fc"
            ]
        },
        "Metadata": {
            "LastTagTime": "0001-01-01T00:00:00Z"
        }
    }
]
```