nextcloud/nextcloudpi #1137: Failed to start The Apache HTTP Server

Oberlurch3000, 2020-05-13:
Hello,

I tried to install Nexcloud on my raspberry pi 4 running octopi. For installation I run this:
wget https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh

During the installation the following error appears causing the installer not to finish:
The following NEW packages will be installed:
  libapache2-mod-authnz-external pwauth
0 upgraded, 2 newly installed, 0 to remove and 44 not upgraded.
Need to get 35.3 kB of archives.
After this operation, 93.2 kB of additional disk space will be used.
Get:1 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian buster/main armhf libapache2-mod-authnz-external armhf 3.3.2-0.1+b5 [24.4 kB]
Get:2 http://mirror1.hs-esslingen.de/pub/Mirrors/archive.raspbian.org/raspbian buster/main armhf pwauth armhf 2.3.11-0.2 [10.9 kB]
Fetched 35.3 kB in 1s (67.8 kB/s)
Selecting previously unselected package libapache2-mod-authnz-external.
(Reading database ... 97651 files and directories currently installed.)
Preparing to unpack .../libapache2-mod-authnz-external_3.3.2-0.1+b5_armhf.deb ...
Unpacking libapache2-mod-authnz-external (3.3.2-0.1+b5) ...
Selecting previously unselected package pwauth.
Preparing to unpack .../pwauth_2.3.11-0.2_armhf.deb ...
Unpacking pwauth (2.3.11-0.2) ...
Setting up libapache2-mod-authnz-external (3.3.2-0.1+b5) ...
apache2_invoke: Enable module authnz_external
Job for apache2.service failed because the control process exited with error code.
See "systemctl status apache2.service" and "journalctl -xe" for details.
invoke-rc.d: initscript apache2, action "restart" failed.
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2020-05-13 19:43:44 BST; 28ms ago
     Docs: https://httpd.apache.org/docs/2.4/
  Process: 1591 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

May 13 19:43:44 octopi systemd[1]: Starting The Apache HTTP Server...
May 13 19:43:44 octopi apachectl[1591]: (98)Address already in use: AH00072: make_sock: could not bind to address [::]:80
May 13 19:43:44 octopi apachectl[1591]: (98)Address already in use: AH00072: make_sock: could not bind to address 0.0.0.0:80
May 13 19:43:44 octopi apachectl[1591]: no listening sockets available, shutting down
May 13 19:43:44 octopi apachectl[1591]: AH00015: Unable to open logs
May 13 19:43:44 octopi apachectl[1591]: Action 'start' failed.
May 13 19:43:44 octopi apachectl[1591]: The Apache error log may have more information.
May 13 19:43:44 octopi systemd[1]: apache2.service: Control process exited, code=exited, status=1/FAILURE
May 13 19:43:44 octopi systemd[1]: apache2.service: Failed with result 'exit-code'.
May 13 19:43:44 octopi systemd[1]: Failed to start The Apache HTTP Server.
Setting up pwauth (2.3.11-0.2) ...
Processing triggers for man-db (2.8.5-2) ...
Module authnz_external already enabled
Module authn_core already enabled
Considering dependency authn_core for auth_basic:
Module authn_core already enabled
Module auth_basic already enabled
Site nextcloud disabled.
To activate the new configuration, you need to run:
  systemctl reload apache2
Enabling site ncp-activation.
To activate the new configuration, you need to run:
  systemctl reload apache2
New password: Retype new password: passwd: password updated successfully
Created symlink /etc/systemd/system/multi-user.target.wants/nextcloud-domain.service → /usr/lib/systemd/system/nextcloud-domain.service.
Downloading updates
Please do not run git as root, your regular user account is enough :)

If you need to run git with root rights for some other application than
what comes preinstalled on this image you can remove this sanity check:

    sudo rm /root/bin/git

You might have to restart your login session after doing that.
No internet connectivity
Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Set an admin username.
Set an admin password.

I dont know what to do with this information or how to fiy that.

ovpc, 2020-05-23:
@Oberlurch3000 
> raspberry pi 4 running octopi. 

The install script is known to work only on Debian/Raspbian 10 (Buster).
No other Os's are supported.
Please use the [forum](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm/33/l/latest) if need assistance. Try the ncp image or use the docker container

Oberlurch3000, 2020-05-23:
SOLVED: The problem was git not working with sudo command. The git version that comes with octopi has it disabled. So I removed it like discribed above. Now it works perfectly on my octopi. ;)

closed 2020-06-13 by nachoparker