nextcloud/nextcloudpi #1141: Default configuration not working on Raspberry Pi 4 with latest Raspberry Pi OS (InnoDB: Cannot allocate memory for the buffer pool)

kb173, 2020-06-08:
On a fresh install of Raspberry Pi OS (version from  2020-05-27) on a new Raspberry Pi 4 with 8 GB RAM, either using Docker or a traditional installation, the Nextcloud server can't fully start. When opening the website at localhost and pressing "Activate", it gets stuck at "NextCloudPi not yet initialized, trying again in a few seconds...".

This is because MariaDB fails to start: `docker logs nextcloudpi` yields the line `Starting MariaDB database server: mysqld . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . failed!`

Furthermore, starting `mysqld` from the Docker container yields this in `/var/log/mysql/error.log`:

```
2020-06-08 22:46:39 0 [Note] InnoDB: Initializing buffer pool, total size = 3.125G, instances = 1, chunk size = 128M
2020-06-08 22:46:39 0 [ERROR] InnoDB: mmap(136118272 bytes) failed; errno 12
2020-06-08 22:46:40 0 [ERROR] InnoDB: Cannot allocate memory for the buffer pool
2020-06-08 22:46:40 0 [ERROR] InnoDB: Plugin initialization aborted with error Generic error
2020-06-08 22:46:40 0 [Note] InnoDB: Starting shutdown...
double free or corruption (out)
200608 22:46:41 [ERROR] mysqld got signal 6 ;
This could be because you hit a bug. It is also possible that this binary
or one of the libraries it was linked against is corrupt, improperly built,
or misconfigured. This error can also be caused by malfunctioning hardware.

To report this bug, see https://mariadb.com/kb/en/reporting-bugs

We will try our best to scrape up some info that will hopefully help
diagnose the problem, but since we have already crashed, 
something is definitely wrong and this may fail.

Server version: 10.3.22-MariaDB-0+deb10u1
key_buffer_size=134217728
read_buffer_size=131072
max_used_connections=0
max_threads=153
thread_count=0
It is possible that mysqld could use up to 
key_buffer_size + (read_buffer_size + sort_buffer_size)*max_threads = 466218 K  bytes of memory
Hope that's ok; if not, decrease some variables in the equation.

Thread pointer: 0x0
Attempting backtrace. You can use the following information to find out
where mysqld died. If you see no messages after this, something went
terribly wrong...
stack_bottom = 0x0 thread_stack 0x49000
The manual page at http://dev.mysql.com/doc/mysql/en/crashing.html contains
information that should help you find out what is causing the crash.
Writing a core file...
Working directory at /data/database
Resource Limits:
Limit                     Soft Limit           Hard Limit           Units     
Max cpu time              unlimited            unlimited            seconds   
Max file size             unlimited            unlimited            bytes     
Max data size             unlimited            unlimited            bytes     
Max stack size            8388608              unlimited            bytes     
Max core file size        unlimited            unlimited            bytes     
Max resident set          unlimited            unlimited            bytes     
Max processes             unlimited            unlimited            processes 
Max open files            1048576              1048576              files     
Max locked memory         65536                65536                bytes     
Max address space         unlimited            unlimited            bytes     
Max file locks            unlimited            unlimited            locks     
Max pending signals       61664                61664                signals   
Max msgqueue size         819200               819200               bytes     
Max nice priority         0                    0                    
Max realtime priority     0                    0                    
Max realtime timeout      unlimited            unlimited            us        
Core pattern: core
```

I could temporarily resolve this be editing `/etc/mysql/mariadb.conf.d/91-npc.cnf` and decreasing the `innodb_buffer_pool_size` from 3341182566 to (e.g.) 341182566 After this, the mysql database starts with `mysqld`. The server is then functional, although very slow - not sure if that's related to this issue.

I've tested this on a completely new Raspberry Pi with a completely fresh Raspberry Pi OS install, so I'm fairly certain that this isn't caused by some configuration error on my side, although I suppose a hardware issue is always possible.

Also, I'm new to this so I'm not sure whether my workaround is sensible - it was just the first thing I tried that worked (it made sense due to the mysql error log). Please let me know if there is a better way to temporarily resolve this.

**System information**

Raspberry Pi 4 with 8 GB RAM

nachoparker, 2020-06-12:
most likely sd card issue or power supply

kb173, 2020-06-19:
I've tried a different power supply as well as a new SD card (different brand too, with a fresh install), and I'm still getting the exact same error :/ As I said, I don't need support, so I don't think the tag is appropriate (I've already fixed it for me locally). It seems like an issue with the default config to me. Unless the Raspberry Pi itself is faulty, but that doesn't seem likely?

yebo29, 2020-07-12:
I have the same issue. Came here to open an issue but it seems this is not the place. I have the offical SD Card and power supply (5V 3 AMPS) from Canakit, so I know that's not it. @kb173 did you post on the forum?

yebo29, 2020-07-12:
Okay for those that have the same problem with a Raspberry Pi4 8GB RAM, this is what I've found works:

1. Modify /data/bin/ncp/CONFIG/nc-limits.sh: Change:
```bash
 # DATABASE MEMORY
  AUTOMEM=$(( TOTAL_MEM * 28 / 100 ))
```
Here we see `AUTOMEM` calculation moves from multiplying by 40 to `28`. At `29` it breaks again.
2. At the end of that file:
```bash
# redis max memory
  local CONF=/etc/redis/redis.conf
  local CURRENT_REDIS_MEM=$( grep "^maxmemory" "$CONF" | awk '{ print $2 }' )
  #[[ "$REDISMEM" != "$CURRENT_REDIS_MEM" ]] && {
    #sed -i "s|^maxmemory .*|maxmemory $REDISMEM|" "$CONF"
  chown redis:redis "$CONF"
  service redis-server restart
  #}
```
Redis memory calculation is broken. It sets `maxmemory` to blank. Just comment out the test.
3. Add `configure` to the end of the file, right after `install() { :; }`
That's what worked for me, but I don't fully understand how the file is called. I added a line to `/etc/services-enabled.d/010lamp` to execute the file just because I saw that's where it bombed out when restarting the container (`
/etc/services-enabled.d/010lamp: line 41:    94 Aborted                 mysqld`):
```bash
# start
#echo "Starting mariaDB"
#mysqld &

# Run nc-limits to fix mysql innodb bug
/data/bin/ncp/CONFIG/nc-limits.sh
```
Note to comment out the previous start lines as nc-limits.sh does that for us.

yebo29, 2020-07-12:
Okay for those that have the same problem with a Raspberry Pi4 8GB RAM, this is what I've found works:

1. Modify /data/bin/ncp/CONFIG/nc-limits.sh: Change:
```bash
 # DATABASE MEMORY
  AUTOMEM=$(( TOTAL_MEM * 28 / 100 ))
```
Here we see `AUTOMEM` calculation moves from multiplying by 40 to `28`. At `29` it breaks again.
2. At the end of that file:
```bash
# redis max memory
 # local CONF=/etc/redis/redis.conf
 # local CURRENT_REDIS_MEM=$( grep "^maxmemory" "$CONF" | awk '{ print $2 }' )
  #[[ "$REDISMEM" != "$CURRENT_REDIS_MEM" ]] && {
    #sed -i "s|^maxmemory .*|maxmemory $REDISMEM|" "$CONF"
    #chown redis:redis "$CONF"
   #service redis-server restart
  #}
```
Redis memory calculation is broken. It sets `maxmemory` to blank. Just comment out the whole block, and modify `maxmemory` in the config file 100mb (all the way at the end of the file). I don't know if that's a sane setting.
3. Add `configure` to the end of the file, right after `install() { :; }`
That's what worked for me, but I don't fully understand how the file is called. I added a line to `/etc/services-enabled.d/010lamp` to execute the file just because I saw that's where it bombed out when restarting the container (`
/etc/services-enabled.d/010lamp: line 41:    94 Aborted                 mysqld`):
```bash
# Run nc-limits to fix mysql innodb bug
/data/bin/ncp/CONFIG/nc-limits.sh

# start
#echo "Starting mariaDB"
service mysql restart
```

I wouldn't recommend ANY of these settings for a production environment, as it's just me experimenting. Also, probably if you update the container or otherwise destroy and recreate it, these settings will be wiped.

nachoparker, 2020-07-29:
since there's a couple people reporting this, let's reopen and retag. I'll ask in the Telegram group if anybody else is seeing this

nachoparker, 2020-07-29:
reopening

nachoparker, 2020-07-29:
reopening

nachoparker, 2020-07-29:
reopening

nachoparker, 2020-07-29:
reopen

nachoparker, 2020-07-29:
@yebo29 thanks for your investigation. What are the values of the AUTOMEM and TOTALMEM variables in your system? also, what is the value of `innodb_buffer_pool_size` in `91-ncp.conf` without any code changes?

Profator, 2020-07-30:
@nachoparker Hello, I have the same Problem. At the moment I temporarily solve it, by manually editing the 91-ncp.conf, changing the innodb_buffer_pool_size to 2G (this is necessary after every restart). I restarted my Pi and the standard is 3303409254. This is also what I get when running nc-limit again. 
How can I check the values of AUTOMEM and TOTALMEM? I can't find them in the ncp-log under /var/logs/ncp.log.
Thanks in advance for helping!

Arrgg, 2020-08-01:
Seeing same behavior on a Pi4/8GB, fresh install.

nachoparker, 2020-08-02:
I don't have a 8GiB handy to test, but I think it should be fixed in `devel`. If anyone can confirm this, I'll merge the fix.

Do 

```
sudo ncp-update devel
```

, and see if it is fixed now

Profator, 2020-08-02:
I just tested it with my pi and it seems to work so far. The calculated value is around the 2G i manually inserted, so it shouldnt be much difference in practice.
@nachoparker Thanks very much for your support! And of course for your excellent work in general!

nachoparker, 2020-08-03:
great! thanks for verifying

closed 2020-08-03 by nachoparker

kb173, 2020-08-03:
Thanks very much @nachoparker!

feigenmann, 2020-10-23:
Hi there I am having the same problem I have an RPI4 8GB and I have innodb_buffer_pool size of 2014299136 and I have also change to mem in ncp-config to 8G
but still i cnat get rid of the activaton screen in nextcloud

which says its not activated yet..

thanks for any help it will be highly appreciated

ovpc, 2020-10-23:
@feigenmann  Please the forum, Github is not for support. Thank you.

feigenmann, 2020-10-23:
@ovpc  which forum - whats the link.

ovpc, 2020-10-23:
https://help.nextcloud.com/tag/ncp

F-Schmalstieg, 2020-12-11:
@nachoparker Just as an Idea, why this problem occurs, could it be, that it is the 32bit-Version of Raspbian? So it is not possible to fully use the 8GB of RAM?
Maybe this could be checked in nc-limits.sh. I will try to test this with the beta 64-bit Version of Raspbian, but at the moment I don't have enough time.
Just as an Idea, because my feeling is, that it cannot be the final solution to just don't use the memory, the pi actually offers. But I don't have enough expertise to validate this theory.

nachoparker, 2020-12-11:
the latest image uses 64 bit OS for raspberry pi