nextcloud/nextcloudpi #1142: Ports show closed in nc-info, but are open.

ovpc, 2020-06-13:
Not really an issue, as ports are in fact open and both NC and NCP web and ncp-config working as expected. Just strange to see ncp reporting closed ports, that are in fact open.
Can someone  look into it and/or explain how this can happen. It is occuring on all my test and production instances. It seems harmless, as it doesnt seem to afect anything, just annoying and confusing. I find the same thing in forum when users include nc-info. Assuming its a bug

Running nc-info
Gathering information...
NextCloudPi version  v1.26.0
NextCloudPi image    NextCloudPi_03-09-19
distribution         Raspbian GNU/Linux 10 \n \l
automount            no
USB devices          sda 
datadir              /media/xnbox/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      11G/932G
rootfs usage         2,7G/15G
swapfile             /dev/zram3
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    18.0.6.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   192.168.2.26
gateway              192.168.2.254
interface            eth0
certificates         
NAT loopback         no
uptime               9days

dl1dby, 2020-06-13:
It'not harmless as after updating to NC 18.05 and NCP v1.26 I cannot upload anything. No Webdav, no Caldav. Makes installation pretty unusable. Downloads, Propfind works though.

suxlex, 2020-06-13:
I have that problem too.

Empfangen: 13.06.2020 08:38:01
Betreff: [nextcloud/nextcloudpi] Ports show closed in nc-info, but are open. (#1142)

nachoparker, 2020-06-13:
you are right. I didn't notice this before. The port checker changed its API URL. Easy fix.

Fixed now.

FadeFx, 2020-09-20:
Seems this issue is still not fixed for me in pannel 1.30 and also previous versions. I dont pay much attention to it however, as actually both ports are reachable in reality and letsencrypt renewal also works without issues.

ovpc, 2020-09-20:
Might be change upstream, it has happened before. Will try to replicate later.

keinhen, 2020-09-27:
Same here, just updated to ncp v1.30.0 and both port check 80 and 443 report closed, even though webUI is still remotely accessible and I can renew my letsencrypt certificates without any issue.

dsmic, 2020-10-09:
I did a little digging: it seems, that portchecker.co changed the API again. Not sure, if they do not like nextcloudpi to use it?!

They generate a _csrf value on the call to https://portchecker.co and generate a corresponding cookie.

So one has to first

wget --keep-session-cookies --save-cookies cookies.txt  https://portchecker.co

to keep the cookie and extract the _csrf value from the index.html file

after this

wget --load-cookies cookies.txt  https://portchecker.co --post-data="port=80&_csrf=XXXXX"

with XXXXX being the correct _scrf value from the first call.

I would try to create a pull request, but I do not know the correct way to handle the tempory cookie file within nextcloudpi.

Maybe someone more familar with nextcloudpi does this within seconds, I would be glad to test:)

ovpc, 2020-10-09:
Thanks for digging and confirming API changed again.
@nachoparker said it was an easy fix when this happened last time. Maybe not so this time....
He will either quick fix this or point us in the right direction, when he has a moment and becomes aware of this. ;-)

dsmic, 2020-10-09:
I played a little bit around, replacing is_port_open() in ncp-diag seems to work:

```
function is_port_open()
{
  PORT=$1
  tmp_file=$(mktemp)
  v=$(wget -T2 -t1 -q --keep-session-cookies --save-cookies $tmp_file https://portchecker.co -O - | grep -oP "_csrf\" value=\"\K.*\"")
  wget -T2 -t1 -q   --load-cookies $tmp_file https://portchecker.co --post-data "port=$PORT&_csrf=${v::-1}" -O - \
  | grep -q '<span class="green">open</span>' && { echo "open"; return 1; }
  echo "closed"
  rm $tmp_file
}
```

nachoparker, 2020-10-20:
works great, thanks!

closed 2020-10-20 by nachoparker

FadeFx, 2020-12-20:
@nachoparker seems the problem has actually inverted, i have moved from my old pi2b to a fresh pi4 4gig and have installed ncp via curl onto radpberryos 64bit, after the first startup i had ncp pannel show the ports as open despite having them actually still forwarded to the pi2. 
Only thing i imagine, if it actually works, could be that it actually got response from the pi2 and thought it was its own.

dsmic, 2020-12-20:
It is properly exactly what you describe. A server is asked to connect to the ports and if it gets response it returns a webpage, indicating that the port is reachable. The server does not care and does not know, if it is the same machine which asked for the check...

svenb1234, 2022-08-09:
Could it be that NCP checks the wrong IP on docker? https://portchecker.co/ shows open for 80 and 443 but NCP claims they are closed.

FadeFx, 2022-08-09:
no, the issue is not only with docker, i have a standard installation and nc pannel show both orts closed as well

svenb1234, 2022-08-09:
Fair enough. So it's still unresolved and general.

svenb1234, 2022-08-17:
Works for me with 1.49.0