nextcloud/nextcloudpi #1143: No update for Nextcloud 8.0.6

llucax, 2020-06-13:
There is a Nextcloud update for 8.0.6 available but it says the NextCloudPi extension is not available for it. Not sure how a patch release can be marked as incompatible.

[image]

**System information**

Using ncp v1.26.0.

Skipping as I don't think it is relevant in this case, but I can upload if needed.

ovpc, 2020-06-13:
https://docs.nextcloudpi.com/en/staying-up-to-date

llucax, 2020-06-13:
Mmm, I see.

> It is highly recommended to use these tools, other methods, including NC's native updater, may or may not work, use at your own risk.

Is there a particular reason why updating via the native NC updater is not recommended? What is NCP doing that NC doesn't (or vice-versa)?

ovpc, 2020-06-13:
Onother question. What is it about; Use forums please, this is not for support,.....people do not understand or choose to ignore?

closed 2020-06-13 by ovpc

llucax, 2020-06-13:
I understand the difference between a support forum and an issue tracker, but sometimes you have to make a call.

This really felt like a bug, because I see no reason why I wouldn't be able to update NC via the official updater. It looked to me like there was some missing update for NCP to report it was compatible with the latest NC.