nextcloud/nextcloudpi #1145: Scheduled system standby and wakeup

adolci, 2020-06-14:
**Description**  

Add the possibility to configure a scheduled standby for the system (for physical devices) with the auto wakeup. 
*Benefits*: 

- reduce noise
- reduce energy consumption
- increases disk and other hardware components life time

**Example**  

A preinstalled system utility is available for this purpose: rtcwake - https://linux.die.net/man/8/rtcwake
On crontab:
`00 22 * * * /usr/sbin/rtcwake -m off -s 39600`
This will shutdown the system (-m off) at 10:00 p.m. every day and boot again at 9 a.m. (-s 39600)

ovpc, 2020-06-14:
Will this actually work on a sbc ? They don't have a bios or clock internally afaik. Nice idea though.

adolci, 2020-06-14:
> Will this actually work on a sbc ? They don't have a bios or clock internally afaik. Nice idea though.

I think it will not work because it need a Real Time Clock device. On man page there is a note that could help: 

> This is normally used like the old apmsleep utility, to wake from a suspend state like ACPI S1 (standby) or S3 (suspend-to-RAM). Most platforms can implement those without analogues of BIOS, APM, or ACPI.

Of course this functionality is limited to some hardware devices

adolci, 2020-06-14:
Pull request: https://github.com/nextcloud/nextcloudpi/pull/1146

Larsene, 2020-07-01:
No way with the raspberry Pi (no internal clock). You can do that - as i do - with a Witty Pi Card : http://www.uugear.com/product/witty-pi-3-realtime-clock-and-power-management-for-raspberry-pi/

Plus, you'll be able to power external disk and Raspberry from a single 12V power block, if you want to.

sunjam, 2020-09-04:
All you will need is to purchase an external real time clock for the Pi.  There are many variations available for $5 - $10 USD that plug into the GPIO pins.  Cheers!