nextcloud/nextcloudpi #1147: Very high HDD start/stop count on Odroid HC2

Skyhooker, 2020-06-15:
Hello.  This is my first issue report on GitHub, and I just joined specifically to make this report.  I hope I do this correctly.

I am running Nextcloudpi, version 1.26.2, on Odroid-HC2, using the image made specifically for that model.  I am unable to post "sudo ncp-report" output because I have been  unable to log in via Putty (connection refused, and no credentials requested).

The issue I am reporting is an extremely high hard drive start/stop count.  I have verified by running SMART tests that it spins up and down an average of around 175 times per day.  It's not an issue of a spindown timer that needs adjustment, but rather that some process is causing the idle drive to spin back up very frequently, and for no apparent reason.

It's not a sync client that is making requests and causing the issue, because I disable the desktop sync client on my computer after its first sync of the day, and no other user is using a sync client.

This concerns me greatly because my drive is only rated for 300,000 lifetime start/stop cycles, and at this rate, it will be the most likely cause of early hard drive failure.  If there is any setting that can be adjusted in the web GUI interface, I have not found it.

Please let me know if any other information is needed.  Thank you for looking into this!

nachoparker, 2020-06-16:
you'll have to replace your HDD

closed 2020-06-16 by nachoparker

Skyhooker, 2020-06-17:
I don't think I was clear, and you seem to have misunderstood me.  My hard drive is only two weeks old.  I don't mean that the SMART start/stop cycle count is already high, but that the count is increasing very rapidly, by around 175 start/stop cycles every day, due to something in the Nextcloudpi system frequently waking up the HDD when it's idle and has spun down.

I recently replaced my hard drive with this new larger one (by a different manufacturer) and I noticed the same thing with the old one, and confirmed it by SMART tests.  The start/stop cycle count increases by over 5000 cycles per month on average.  If this continues, it is very likely to cause early hard drive failure.

I don't see how this is a forum-support issue, because there is no user-adjustable setting available in any menu to adjust the frequency with which the operating system accesses the HDD when no user is actively using the system.  It seems clear to me that there is an underlying process, not available to the user/admin, that is accessing the HDD frequently, causing it to spin back up from idle.

The issue also can't be caused by the desktop sync client, because I disable that after the first sync of the day.

I do not mean to be annoying, and I really appreciate the work that has gone into this.  I think that this issue is serious enough that it was worth reporting, because it will very likely cause early hardware failure for some users, similar to the very high failure rate of so-called "green" hard drives, due to a high number of start/stop cycles.  I joined Github solely to report this issue.

Thank you.