nextcloud/nextcloudpi #1149: Calender could not be downloaded error

gbafana, 2020-06-23:
I am trying to install nextcloudpi on docker in OMV5 on Odroid HC1(Xu4). I tried several time but i end up at the same following stage.

`Initializing empty volume..
Making /usr/local/etc/ncp-config.d persistent ...
Making /etc/services-enabled.d persistent ...
Making /etc/letsencrypt persistent ...
Making /etc/shadow persistent ...
Making /etc/cron.d persistent ...
Making /etc/cron.daily persistent ...
Making /etc/cron.hourly persistent ...
Making /etc/cron.weekly persistent ...
Making /usr/local/bin persistent ...
Making /etc/apache2 persistent ...
Starting PHP-fpm
Starting Apache
Starting mariaDB
2020-06-23  8:27:07 0 [Note] mysqld (mysqld 10.3.22-MariaDB-0+deb10u1) starting as process 201 ...
Starting Redis
Starting Cron
Starting Postfix
Uninitialized instance, running nc-init...
Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Nextcloud was successfully installed
Set mode for background jobs to 'cron'
System config value tempdirectory set to string /var/www/nextcloud/data/tmp
System config value mysql.utf8mb4 set to boolean true
System config value trusted_domains => 11 set to string <MY IP>
System config value trusted_domains => 1 set to string 172.17.0.4
System config value trusted_domains => 5 set to string nextcloudpi.local
System config value trusted_domains => 7 set to string nextcloudpi
System config value trusted_domains => 8 set to string nextcloudpi.lan
System config value mail_smtpmode set to string sendmail
System config value mail_smtpauthtype set to string LOGIN
System config value mail_from_address set to string admin
System config value mail_domain set to string ownyourbits.com
nextcloudpi enabled
Error: Could not download app calendar
Init done`

I tried it several time but no luck.
Can you help me?

iamABH, 2020-07-01:
Same error and same logs with Raspbery Pi 4

keanecodes, 2020-07-14:
I had the same issue but ignored that error and proceeded anyway. Eventually that error came back to bite me and I got [this error](https://help.nextcloud.com/t/nextcloudpi-nothing-happens-after-initializing-nextcloudpi-for-the-first-time/37898). The thread suggests to clear the `ncdata` volume, which lead me to this [other thread](https://help.nextcloud.com/t/nextcloudpi-how-to-start-with-a-fresh-container/30058/3) and that guided me to run `docker volume ncdata` to clear the volume.

In my case, however, I removed all my docker containers and images just to be sure by following the instructions [here](https://linuxize.com/post/how-to-remove-docker-images-containers-volumes-and-networks/#:~:text=Removing%20Docker%20Volumes,-Removing%20one%20or&text=To%20remove%20one%20or%20more%20Docker%20volumes%20use%20the%20docker,volumes%20you%20want%20to%20remove.&text=If%20you%20get%20an%20error%20similar%20to%20the%20one%20shown,to%20remove%20the%20container%20first.).

Good luck.

closed 2020-08-13 by ovpc