nextcloud/nextcloudpi #1154: Verbose NCP Panel commands

3tyh86qvt, 2020-07-11:
**Description**  
NCP Panel is a marvel for newbies. You can activate or deactivate anything and everything with a couple clicks.

The disadvantage of this simplicity is that newbies are not given any hint about how the commands and settings are activated in the wild terminal, thus hindering their development to autonomous Nextcloud-administrators.

You click on make a backup and the only information that you get is the feedback from the terminal once the command has been activated. We cannot see which terminal command could activate this function without the need to activate NCP Panel.

"[ nc-backup ] (Sat Jul 11 17:34:41 BST 2020)
check free space...
Maintenance mode enabled
backup database...
backup files..."

This way, if you see yourself logging into the NCP Panel and seeing how that command is activated, you might be able to set a cron job or something. 

I know that I cold go to the documentation and look that up, but this way, the system would be showing me how to do it by myself every time I click on the setting.

**Example**  
When you enter the option "backup", for example, either the description tells you which commands will be activated, or else the terminal feedback/output window shows you a "bash" (?) as the commands are being sent to the terminal.

theCalcaholic, 2020-07-11:
That's not so straightforward, actually. The "commands" that get activated are sometimes quite complex and advanced bash scripts.

However, it would be trivial to set the shell option `-x` to show all commands that get executed. However, I don't really see how that would be helpful to anyone who isn't capable of opening and reading the script itself...

3tyh86qvt, 2020-07-12:
They might not be able today, but they will learn if they have interest in it and they will get acquaintance with those commands.
I'm a language teacher. Sometimes, I sometimes speak my students using words they have never heard before. I really see how that is helpful to anyone who isn't capable of going to the country and speaking to the natives themselves.

goebbe, 2021-02-15:
I agree that it would be very helpful to at least provide some information about the implementation. 
Something like: 
"Add a cronjob for user ncp to run /usr/local/bin/auto_backup.sh.  Backups run at 3 a.m."  :-)

nachoparker, 2021-10-14:
I'll just keep in mind the suggestion to add more verbose output to each ncp-app. Thanks