nextcloud/nextcloudpi #1155: AH00037: Symbolic link not allowed or link target not accessible

gustafssone, 2020-07-15:
Fresh installation of Docker instance of nextcloudpi. Using Docker swarm + Traefik (but if I disable Traefik and use 80/443 directly it doesn't make a difference) over port 80. Tried both port 80 and 443 (and Enable/Disabled Force HTTPS). Irrelevant of what I tried, I had the following error in the Apache log:
[Wed Jul 15 08:30:52.703025 2020] [core:error] [pid 19188:tid 2700080160] [client 172.16.200.3:55762] AH00037: Symbolic link not allowed or link target not accessible: /var/www/nextcloud

I tried importing my previous (non-docker) instance of NextcloudPi to see if it reset the permissions, but no difference.

Got a Forbidden error in my browser.
Tried to change rights with chown all over the place, no luck. Even ownership at /mnt/gfsshare/volumes/nextcloudpi to www-data:www-data (see .yml below), but no difference. 

Only solution I could find (which worked), was to change the apache site conf. "Require all granted" had to be there if using another DocumentRoot. Equivalent conf for 443. 
<VirtualHost _default_:80>
  DocumentRoot /data/nextcloud
    CustomLog /var/log/apache2/nc-access.log combined
    ErrorLog  /var/log/apache2/nc-error.log
  <Directory /data/nextcloud/>
    Options +FollowSymlinks
    AllowOverride All
    Require all granted
    <IfModule mod_dav.c>
      Dav off
    </IfModule>
  </Directory>
</VirtualHost>

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.26.2
NextCloudPi image    NextCloudPi_docker_04-06-20
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      88G/1.9T
rootfs usage         7.1G/59G
swapfile             /var/swap
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    18.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth2
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               3days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "18.0.5.1",
        "overwrite.cli.url": "https:\/\/egnholding.se\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/media\/USBdrive\/ncdata\/tmp",
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "maintenance": false,
        "logfile": "\/media\/USBdrive\/ncdata\/nextcloud.log",
        "loglevel": "2",
        "log_type": "file",
        "htaccess.RewriteBase": "\/",
        "mail_sendmailmode": "smtp",
        "mail_smtpsecure": "tls",
        "mail_smtpauth": 1,
        "mail_smtphost": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpport": "587",
        "mail_smtpname": "***REMOVED SENSITIVE VALUE***",
        "mail_smtppassword": "***REMOVED SENSITIVE VALUE***",
        "has_rebuilt_cache": true,
        "twofactor_enforced": "true",
        "twofactor_enforced_groups": [],
        "twofactor_enforced_excluded_groups": [
            "admin"
        ],
        "theme": "",
        "app_install_overwrite": [
            "gallery",
            "previewgenerator"
        ],
        "data-fingerprint": "xx"
    }
}

```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Wed Jul 15 08:30:52.703025 2020] [core:error] [pid 19188:tid 2700080160] [client 172.16.200.3:55762] AH00037: Symbolic link not allowed or link target not accessible: /var/www/nextcloud

```
</details>

<details>
<summary>Database logs</summary>

```

2020-07-15 10:27:09 0 [Note] InnoDB: 10.3.22 started; log sequence number 513358967; transaction id 29955
2020-07-15 10:27:09 0 [Note] Plugin 'FEEDBACK' is disabled.
2020-07-15 10:27:09 0 [Note] InnoDB: Loading buffer pool(s) from /data/database/ib_buffer_pool
2020-07-15 10:27:09 0 [Note] Server socket created on IP: '127.0.0.1'.
2020-07-15 10:27:10 0 [Note] Reading of all Master_info entries succeeded
2020-07-15 10:27:10 0 [Note] Added new Master_info '' to hash table
2020-07-15 10:27:10 0 [Note] mysqld: ready for connections.
Version: '10.3.22-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 10
2020-07-15 10:27:14 0 [Note] InnoDB: Buffer pool(s) load completed at 200715 10:27:14
2020-07-15 10:29:33 0 [Note] mysqld (initiated by: root[root] @ localhost []): Normal shutdown
2020-07-15 10:29:33 0 [Note] Event Scheduler: Purging the queue. 0 events
2020-07-15 10:29:33 0 [Note] InnoDB: FTS optimize thread exiting.
2020-07-15 10:29:33 0 [Note] InnoDB: Starting shutdown...
2020-07-15 10:29:33 0 [Note] InnoDB: Dumping buffer pool(s) to /data/database/ib_buffer_pool
2020-07-15 10:29:33 0 [Note] InnoDB: Instance 0, restricted to 4095 pages due to innodb_buf_pool_dump_pct=25
2020-07-15 10:29:33 0 [Note] InnoDB: Buffer pool(s) dump completed at 200715 10:29:33
2020-07-15 10:29:36 0 [Note] InnoDB: Shutdown completed; log sequence number 513391831; transaction id 30068
2020-07-15 10:29:36 0 [Note] InnoDB: Removed temporary tablespace data file: "ibtmp1"
2020-07-15 10:29:36 0 [Note] mysqld: Shutdown complete

```
</details>

nextcloudpi.yml:

version: "3.7"

services:
  ncp:
    image: ownyourbits/nextcloudpi-armhf
    command: "mydomain.com"
    ports:
     - "80:80" 
     - "443:443"
     - "4443:4443"
    networks:
      - network_traefik_public
    volumes:
     - /mnt/gfsshare/volumes/nextcloudpi:/data 
     - /etc/localtime:/etc/localtime:ro
     - /media/USBdrive:/media/USBdrive
    deploy:
      restart_policy:
        condition: on-failure
        delay: 5s
        max_attempts: 3
        window: 120s

networks:
  network_traefik_public:
      external: true

Docker info:
Server: Docker Engine - Community
 Engine:
  Version:          19.03.12
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.13.10
  Git commit:       48a6621
  Built:            Mon Jun 22 15:47:34 2020
  OS/Arch:          linux/arm
  Experimental:     true
 containerd:
  Version:          1.2.13
  GitCommit:        7ad184331fa3e55e52b890ea95e65ba581ae3429
 runc:
  Version:          1.0.0-rc10
  GitCommit:        dc9208a3303feef5b3839f4323d9beb36df0a9dd
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683

closed 2020-07-15 by ovpc

ovpc, 2020-07-15:
Unless you're reporting a reproducible bug, closed this for now. There are threads on NCP with Traefik, might have to search a bit tho.

gustafssone, 2020-07-15:
@ovpc this is a reproducible bug. I could reproduce it every time. I already found a workaround (or the cause). Install the latest version of nextcloudpi with docker and try to access it. 
I wonder - could it have something to do with glusterfs being used? "/mnt/gfsshare/volumes/nextcloudpi" is based on a glusterfs mount (/mnt/gfsshare/)