nextcloud/nextcloudpi #1156: ncp-restore restores wrong tmp directory

theCalcaholic, 2020-07-19:
When performing `ncp-restore` while having the data directory in a non-standard location, the tmp directory for php-fpm will be set to /var/www/nextcloud/data nonetheless.

This happened on two different instances during an automatic restore after a failed NC upgrade - so it's possible that the bug doesn't affect manual restores.

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.26.2
NextCloudPi image    NextCloudPi_12-21-19
distribution         Armbian 19.11.4 Buster \l 
automount            yes
USB devices          sdb 
datadir              /media/cloud/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      355G/586G
rootfs usage         25G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    18.0.5.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               15:27

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.30.20",
            "7": "nextcloudpi",
            "8": "nextcloudpi.lan",
            "2": "cloud.***REDACTED***.org",
            "11": "149.233.214.148",
            "12": "cloud.***REDACTED***.org",
            "13": "***REDACTED***.org,ice.***REDACTED***.org,pihole.***REDACTED***.org,redmine.***REDACTED***.org
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/cloud.***REDACTED***.org\/",
        "dbtype": "mysql",
        "version": "18.0.5.1",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
[... 58 lines cut ...]
```

{"reqId":"XxOBSJ7gjaRG9AUoaevgPgAAAIs","level":3,"time":"2020-07-18T23:10:01+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/9ab53fc414daa861cfaaf560373a9e71/0000000000000000-0000000002728397","message":"fread(): file created in the system's temporary directory at /var/www/nextcloud/3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"XxOBUJ7gjaRG9AUoaevgVAAAAI8","level":3,"time":"2020-07-18T23:10:10+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/84309a0ecd477c7c1647cdc226640588/0000000000000000-0000000005172919","message":"fread(): file created in the system's temporary directory at /var/www/nextcloud/3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"XxOBd57gjaRG9AUoaevgcQAAAIY","level":3,"time":"2020-07-18T23:10:48+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/59c0abcb76c2eb56890506bc3d3900a8/0000000000000000-0000000005282059","message":"fread(): file created in the system's temporary directory at /var/www/nextcloud/3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"XxOBgPlbv2OErjrDCusF4gAAAA0","level":3,"time":"2020-07-18T23:10:57+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/cf5271466d80831ccb233cd52e6751d2/0000000000000000-0000000004268238","message":"fread(): file created in the system's temporary directory at /var/www/nextcloud/3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"XxOBiJ7gjaRG9AUoaevgiwAAAJg","level":3,"time":"2020-07-18T23:11:06+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/255f1abeed65409bcc14a8aea9ec36b4/0000000000000000-0000000005285696","message":"fread(): file created in the system's temporary directory at /var/www/nextcloud/3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"XxOBk-lbv2OErjrDCusGBQAAAAE","level":3,"time":"2020-07-18T23:11:16+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/b6a1531e0b7ebc9ab613724987e574b7/0000000000000000-0000000005360879","message":"fread(): file created in the system's temporary directory at /var/www/nextcloud/3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"XxOBnJ7gjaRG9AUoaevgtQAAAI0","level":3,"time":"2020-07-18T23:11:25+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/93623922217eb7bf778c1406e90df967/0000000000000000-0000000006067575","message":"fread(): file created in the system's temporary directory at /var/www/nextcloud/3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"XxOBpflbv2OErjrDCusGDwAAAAo","level":3,"time":"2020-07-18T23:11:34+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/b6e50eed7dea54895a0b393c589176da/0000000000000000-0000000005838763","message":"fread(): file created in the system's temporary directory at /var/www/nextcloud/3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"XxOBr57gjaRG9AUoaevg4QAAAIE","level":3,"time":"2020-07-18T23:11:44+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/7eb02d1c26b951a5c191293140c3d7e7/0000000000000000-0000000005561982","message":"fread(): file created in the system's temporary directory at /var/www/nextcloud/3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"XxOBuJ7gjaRG9AUoaevg9QAAAI8","level":3,"time":"2020-07-18T23:11:54+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/6a87f87582f4ff01603a900e2bc085e3/0000000000000000-0000000004490425","message":"fread(): file created in the system's temporary directory at /var/www/nextcloud/3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"XxOBwZ7gjaRG9AUoaevhBwAAAJI","level":3,"time":"2020-07-18T23:12:02+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/9df9d4a395ac8b6828d01a49c84eac5e/0000000000000000-0000000004159145","message":"fread(): file created in the system's temporary directory at /var/www/nextcloud/3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"XxOByoTJ8nNIV7BAmjOWjQAAAFQ","level":3,"time":"2020-07-18T23:12:11+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/3988d8f0bdb341582c5b63b90ff7e342/0000000000000000-0000000010239999","message":"fread(): file created in the system's temporary directory at /var/www/nextcloud/3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"XxOBzoTJ8nNIV7BAmjOWjwAAAEU","level":3,"time":"2020-07-18T23:12:15+00:00","remoteAddr":"149.233.214.148","user":"theCalcaholic","app":"PHP","method":"PUT","url":"/remote.php/dav/uploads/theCalcaholic/3988d8f0bdb341582c5b63b90ff7e342/000 [... cut ...] /3rdparty/icewind/streams/src/Wrapper.php#91","userAgent":"Mozilla/5.0 (Android) Nextcloud-android/3.12.1","version":"18.0.5.1"}
{"reqId":"Eh6JoHxyxYUYf9G6Ko3o","level":3,"time":"2020-07-19T02:03:27+00:00","remoteAddr":"","user":"--","app":"PHP","method":"","url":"--","message":"unlink(/media/cloud/ncdata/appdata_ocq4mkumircu/preview/868198/256-561.jpg): No such file or directory at /var/www/nextcloud/lib/private/Files/Storage/Local.php#229","userAgent":"--","version":"18.0.5.1"}

```
</details>

ldw12, 2020-09-02:
experience the same thing after a nextcloud update from 18.0.7 to 19.0.2.
Had to manually change the "upload_tmp_dir" and "sys_temp_dir" field to my external disk's tmp data directory in /etc/php/7.3/fpm/php.ini file for it to work. Else, I won't be able to upload files to the server, as the "/var/www/nextcloud/data/tmp" directory doesn't exist.

nachoparker, 2020-09-12:
I don't see how this can happen. I imagine an `apt-get update` overwrote this

https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp/BACKUPS/nc-restore.sh#L144

Also we are now making sure that this doesn't happen automatically during ncp updates

https://github.com/nextcloud/nextcloudpi/commit/3cf269a305082fde9989df6c485132a2d5bb98e1#diff-9d93438b930005f9ecf7cef69cfb298dR299

So I think we are safe to close this for now

closed 2020-09-12 by nachoparker

foocp, 2020-12-30:
It just happened to me, after a failed upgrade from 19 to 20. (failed because of my mistake, btw). Which led to an error that possibly many face at some point. See [here](https://github.com/nextcloud/server/issues/22509#issuecomment-748911985)