nextcloud/nextcloudpi #1159: NextCloud's Admin overview complains missing php-imagick

ptandler, 2020-07-25:
I just installed a NextCloudPi
One thing I noticed that on https://nextcloudpi.local/settings/admin/overview

The newly installed NextCloud complains about missing PHP module `imagick`.

It could be easily fixed by doing `sudo apt install php-imagick`. However, I guess this should be part of the NCP image?

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v1.28.0
NextCloudPi image    NextCloudPi_03-28-20
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          sda sdb 
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      773M/1.9T
rootfs usage         2.2G/29G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    18.0.7.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               56min

```
</details>

mad73923, 2020-07-25:
Imagick isn't installed deliberately due to security concerns: https://github.com/nextcloud/server/issues/13099#issue

closed 2020-07-25 by ovpc

ptandler, 2020-07-26:
Ah, I see. Thanks for the explanation.