nextcloud/nextcloudpi #1161: HDD Spindown Control

JellyFisch, 2020-07-27:
( sorry it is my first created ticket be lenient i am open for constructive criticism )

Can you add the possibility to configure a scheduled HDD Spindown
The Benefits:

-reduce noise
-reduce energy consumption
-increases disk life time

**Example**  
one Tool is hdparm (  there are 2 other programs known to me that can do this hd-idle and sdparm )

edit the configuration file to make hdparm configuration permanent 

sudo nano /etc/hdparm.conf

The spindown_time value is multiplied by 5 and you have the total time in seconds. So a value of 120 yields 10 minutes (120*5=600).

sample Entry:

/dev/sda {
write_cache = on
spindown_time = 120
}

ovpc, 2020-08-13:
We have no manpower to investigate or implement this, so unless you are going to create a PR yourself. Also I do not see anyone else seconding this feature request.

svenb1234, 2020-09-06:
Is this not a duplicate of https://github.com/nextcloud/nextcloudpi/issues/1145 ?

Tesch1933, 2021-04-25:
I also think a spindown function would be very useful.  Especially for everyone.  who do not access their Nextcloud every day.

dotlineX, 2022-01-22:
I am using hd-idle for this. Works great

WeiChihChern, 2022-01-22:
@dotlineX Would you please elaborate on how to use it with Nextcloud? Thanks

dotlineX, 2022-01-24:
> @dotlineX Would you please elaborate on how to use it with Nextcloud? Thanks

Of course! However, I don't remember exactly how I set it up but it was very basic. 
As far as I know, it is a systemctl service running in the background that checks for disk activity. In my setup, if it hasn't seen any activity for 600 seconds (default, I think) it spins down the drive. The moment you (or nextcloud or any other service) tries to access it, the drive spins up again.

It's a very basic tool to be honest. The config file is in /etc/default/hd-idle. It can also log spinup and spindown times.

I hope that helps.

WeiChihChern, 2022-01-24:
@dotlineX Thank you for the explanation. Will look into it!

steviehs, 2022-07-25:
using hd-idle is from what I know also the most efficient way to spin down a hdd as hdparm is not supported by all disks (I remember some wd drives). funny enough hdparm still works sometimes to detect the state of a drive.
The biggest PITA in that situation are USB3 HDDs which do not support spin down at all... 
I have added support for hd-idle in https://gitlab.com/syncosync/syncosync/-/blob/master/src/soscore/drivemanager.py maybe this would help making a pr here?
For getting the quite cheap and low powered Seagate Expansion 2.5" USB Drive to work, adding 
sed -i '$s/$/ usb-storage.quirks=0bc2:ac25:/' /boot/cmdline.txt
is necessary on the RPi.