nextcloud/nextcloudpi #1162: Debian buster Curl install

fmax, 2020-07-29:
it seems to be the same problem again, as one year ago: https://github.com/nextcloud/nextcloudpi/issues/976

tested on raspberry4 with OS debian buster and virtualbox debian buster 64bit with resulting in the same problem.

running curl install `curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash `
ends up with following output:
```
Installing nc-ramlogs
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   121  100   121    0     0    420      0 --:--:-- --:--:-- --:--:--   420
100  7101    0  7101    0     0   7752      0 --:--:-- --:--:-- --:--:-- 12015
#####         Reboot to activate log2ram         #####
##### edit /etc/log2ram.conf to configure options ####
rm: cannot remove '/etc/cron.daily/log2ram': No such file or directory
Running nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Nextcloud was successfully installed
Set mode for background jobs to 'cron'
System config value tempdirectory set to string /var/www/nextcloud/data/tmp
System config value mysql.utf8mb4 set to boolean true
System config value trusted_domains => 11 set to string 178.165.130.44
System config value trusted_domains => 1 set to string 10.0.2.15
System config value trusted_domains => 5 set to string nextcloudpi.local
System config value trusted_domains => 7 set to string nextcloudpi
System config value trusted_domains => 8 set to string nextcloudpi.lan
System config value mail_smtpmode set to string sendmail
System config value mail_smtpauthtype set to string LOGIN
System config value mail_from_address set to string admin
System config value mail_domain set to string ownyourbits.com
cp: cannot stat '/var/www/ncp-app': No such file or directory
```
full output pastebin: https://pastebin.com/VF8QimGV

additional info:
on friday, 24th july 2020 the installer was working
the last commits of this git repos have no changes causing this issue.
so i think, some of upstream content used by this repos has changed.

ovpc, 2020-07-29:
Thanks for reporting, upstream changes has been known to cause breaking curl installer.
I will try in a VM, to see if I can reproduce it.

fmax, 2020-07-29:
as waiting for the external bugfix, you can use following workaround:
```
apt install -y unzip rsync

mkdir /root/temp/
cd /root/temp/

wget https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh
chmod 0770 install.sh

wget https://github.com/nextcloud/nextcloudpi/archive/master.zip
unzip master.zip

mkdir -p /var/www/ncp-app/ /var/www/ncp-web/ /var/www/ncp-previewgenerator

chown root:root /var/www/ncp-app/
chmod 0755 /var/www/ncp-app/

chown www-data:www-data /var/www/ncp-web/
chmod 0770 /var/www/ncp-web/

chown www-data:www-data /var/www/ncp-previewgenerator
chmod 0770 /var/www/ncp-previewgenerator

rsync -avH nextcloudpi-master/ncp-app/ /var/www/ncp-app/
rsync -avH nextcloudpi-master/ncp-web/ /var/www/ncp-web/
rsync -avH nextcloudpi-master/ncp-previewgenerator/ /var/www/ncp-previewgenerator/

mkdir -p /usr/local/etc/ncp-config.d/

chgrp -R www-data /usr/local/etc/ncp-config.d/
chmod -R 2755 /usr/local/etc/ncp-config.d/

touch /usr/local/etc/ncp-config.d/nc-admin.cfg
touch /usr/local/etc/ncp-config.d/nc-passwd.cfg

chmod 0660 /usr/local/etc/ncp-config.d/nc-admin.cfg
chmod 0660 /usr/local/etc/ncp-config.d/nc-passwd.cfg

./install.sh
```

daschmidt1994, 2020-07-30:
Same probleme here.

PepperPhil, 2020-08-01:
> as waiting for the external bugfix, you can use following workaround:
>
> ```
> apt install -y unzip rsync
>
> mkdir /root/temp/
> cd /root/temp/
>
> wget [...]

Many thanks ... that worked for me.

jbruck, 2020-08-02:
I found issues on fresh install of Debian Buster running on i3 NUC.
Running fmax's instructions allowed the install to finish but the ncp-web gives failed authorization errors and as such the first run wizard does not work.

nachoparker, 2020-08-02:
Should be fixed now

closed 2020-08-02 by nachoparker