nextcloud/nextcloudpi #1165: Nextcloupi on Raspberry 4 8GB-RAM-Version

praktikant-pi, 2020-08-12:
Hello NextCloudPi Community.
I am new at GitHub - so sorry if i break some code of conduct. I tried to install ncp on my new Raspberry pi 4 8GB RAM-Version. It didn't work via the curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/ins... on a fresh installed 32-raspbian OS or on a fresh installed 64-beta raspbian OS or with the ready to use 32-bit ncp image. I found some forums and all 8GB-RAM Guys are having the same Problem. Don't now which information you need from me. But i will happily hand them in later if you demand.
Thanks a lot in anticipation

**System information**

ovpc, 2020-08-13:
Please use the forum for help, if it turns out to be a bug we can reopen here. You are not providing any system info or logs.
Maybe this thread will be useful https://help.nextcloud.com/t/cannot-allocate-memory-for-bufferpool-mariadb-on-fresh-raspberry-pi-4-8gb/87067/6 
If not create one for your issue and add output of ncp-report via pastebin or similar.

closed 2020-08-13 by ovpc

nachoparker, 2020-08-13:
FYI the 8GB version is reported to work fine after the latest fixes

https://github.com/nextcloud/nextcloudpi/issues/1141