nextcloud/nextcloudpi #1166: Create storage failed (External Storage DropBox)

Grizaster, 2020-08-25:
I hope I do this right. I have a problem with the NextCloudApp External Storage DropBox.
Everything looks fine but I get an Exception (see attached image).

**System information**

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v1.25.0
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          sda
datadir              /media/myCloudDrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      78G/466G
rootfs usage         4.2G/118G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    18.0.4.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         yes
uptime               8:36

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

Cannot load Zend OPcache - it was already loaded
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.178.203",
            "11": "79.199.46.90",
            "5": "pinextcloud.local",
            "7": "pinextcloud",
            "8": "pinextcloud.lan",
            "2": "ncp.benibensn.de",
            "6": "pinextcloud.fritz.box"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "18.0.4.2",
        "overwrite.cli.url": "https:\/\/ncp.benibensn.de\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
[... 98 lines cut ...]

{"reqId":"X0UQ5HVa3Kptx1RDUR9BnAAASww","level":3,"time":"2020-08-25T13:23:48+00:00","remoteAddr":"93.240.156.150","user":"bensn","app":"files_external","method":"DELETE","url":"/apps/files_external/userstorages/3","message":{"Exception":"Exception","Message":"Creating \\OCA\\Files_external_dropbox\\Storage\\Dropbox storage failed","Code":0,"Trace":[{"file":"/var/www/nextcloud/apps/files_external/lib/Service/StoragesService.php","line":516,"function":"__construct","class":"OCA\\Files_external_dropbox\\Storage\\Dropbox","type":"->","args":[{"configured":"false","client_id":"xzkb7k5zybokmnl","client_secret":"cywf2jp9esvugwq","token":"false"}]},{"file":"/var/www/nextcloud/apps/files_external/lib/Service/StoragesService.php","line":482,"function":"getRustyStorageIdFromConfig","class":"OCA\\Files_External\\Service\\StoragesService","type":"->","args":[{"__class__":"OCA\\Files_External\\Lib\\StorageConfig"}]},{"file":"/var/www/nextcloud/apps/files_external/lib/Controller/StoragesController.php","line":355,"function":"removeStorage","class":"OCA\\Files_External\\Service\\StoragesService","type":"->","args":["3"]},{"file":"/var/www/nextcloud/apps/files_external/lib/Controller/UserStoragesController.php","line":225,"function":"destroy","class":"OCA\\Files_External\\Controller\\StoragesController","type":"->","args":["3"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":170,"function":"destroy","class":"OCA\\Files_External\\Controller\\UserStoragesController","type":"->","args":["3"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":99,"function":"executeController","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->","args":[{"__class__":"OCA\\Files_External\\Controller\\UserStoragesController"},"destroy"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":125,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->","args":[{"__class__":"OCA\\Files_External\\Controller\\UserStoragesController"},"destroy"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Routing/RouteActionHandler.php","line":47,"function":"main","class":"OC\\AppFramework\\App","type":"::","args":["OCA\\Files_External\\Controller\\UserStoragesController","destroy",{"__class__":"OC\\AppFramework\\DependencyInjection\\DIContainer"},{"id":"3","_route":"files_external.user_storages.destroy"}]},{"function":"__invoke","class":"OC\\AppFramework\\Routing\\RouteActionHandler","type":"->","args":[{"id":"3","_route":"files_external.user_storages.destroy"}]},{"file":"/var/www/nextcloud/lib/private/Route/Router.php","line":299,"function":"call_user_func","args":[{"__class__":"OC\\AppFramework\\Routing\\RouteActionHandler"},{"id":"3","_route":"files_external.user_storages.destroy"}]},{"file":"/var/www/nextcloud/lib/base.php","line":1008,"function":"match","class":"OC\\Route\\Router","type":"->","args":["/apps/files_external/userstorages/3"]},{"file":"/var/www/nextcloud/index.php","line":38,"function":"handleRequest","class":"OC","type":"::","args":[]}],"File":"/var/www/nextcloud/apps/files_external_dropbox/lib/Storage/Dropbox.php","Line":104,"CustomMessage":"--"},"userAgent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:79.0) Gecko/20100101 Firefox/79.0","version":"18.0.4.2"}
{"reqId":"X0UQ8HVa3Kptx1RDUR9B8AAAUhM","level":3,"time":"2020-08-25T13:24:00+00:00","remoteAddr":"93.240.156.150","user":"bensn","app":"core","method":"GET","url":"/core/preview?fileId=142979&x=32&y=32","message":{"Exception":"ImagickException","Message":"attempt to perform an operation not allowed by the security policy `PDF' @ error/constitute.c/IsCoderAuthorized/408","Code":499,"Trace":[{"file":"/var/www/nextcloud/lib/private/Preview/Bitmap.php","line":88,"function":"readimage","class":"Imagick","type":"->","args":["/media/myCloudDrive/ncdata/bensn/files/Documents/Hausbau/Plan/Werkplan Schnitt.pdf[0]"]},{"file":"/var/www/nextcloud/lib/private/Preview/Bitmap.php","line":51,"function":"getResizedPreview","class":"OC\\Preview\\Bitmap","type":"->","args":["/media/myCloudDrive/ncdata/bensn/files/Documents/Hausbau/Plan/Werkplan Schnitt.pdf",2048,2048]},{"file":"/var/www/nextcloud/lib/private/Preview/GeneratorHelper.php","line":62,"function":"getThumbnail","class":"OC\\Preview\\Bitmap","type":"->","args":[{"__class__":"OC\\Files\\Node\\File"},2048,2048]},{"file":"/var/www/nextcloud/lib/private/Preview/Generator.php","line":205,"function":"getThumbnail","class":"OC\\Preview\\GeneratorHelper","type":"->","args":[{"__class__":"OC\\Preview\\PDF"},{"__class__":"OC\\Files\\Node\\File"},2048,2048]},{"file":"/var/www/nextcloud/lib/private/Preview/Generator.php","line":127,"function":"getMaxPreview","class":"OC\\Preview\\Generator","type":"->","args":[{"__class__":"OC\\Files\\SimpleFS\\SimpleFolder"},{"__class__":"OC\\Files\\Node\\File"},"application/pdf",""]},{"file":"/var/www/nextcloud/lib/private/PreviewManager.php","line":185,"function":"getPreview","class":"OC\\Preview\\Generator","type":"->","args":[{"__class__":"OC\\Files\\Node\\File"},32,32,true,"fill","application/pdf"]},{"file":"/var/www/nextcloud/core/Controller/PreviewController.php","line":172,"function":"getPreview","class":"OC\\PreviewManager","type":"->","args":[{"__class__":"OC\\Files\\Node\\File"},32,32,true,"fill"]},{"file":"/var/www/nextcloud/core/Controller/PreviewController.php","line":144,"function":"fetchPreview","class":"OC\\Core\\Controller\\PreviewController","type":"->","args":[{"__class__":"OC\\Files\\Node\\File"},32,32,false,true,"fill"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":170,"function":"getPreviewByFileId","class":"OC\\Core\\Controller\\PreviewController","type":"->","args":[142979,32,32,false,true,"fill"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Http/Dispatcher.php","line":99,"function":"executeController","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->","args":[{"__class__":"OC\\Core\\Controller\\PreviewController"},"getPreviewByFileId"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/App.php","line":125,"function":"dispatch","class":"OC\\AppFramework\\Http\\Dispatcher","type":"->","args":[{"__class__":"OC\\Core\\Controller\\PreviewController"},"getPreviewByFileId"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Routing/RouteActionHandler.php","line":47,"function":"main","class":"OC\\AppFramework\\App","type":"::","args":["OC\\Core\\Controller\\PreviewControl [... cut ...] .0.4.2"}
{"reqId":"4McvKtuRo23wTxLXuSZf","level":2,"time":"2020-08-25T13:35:05+00:00","remoteAddr":"","user":"--","app":"mail","method":"","url":"--","message":"not enough messages to train a classifier","userAgent":"--","version":"18.0.4.2"}
{"reqId":"4McvKtuRo23wTxLXuSZf","level":2,"time":"2020-08-25T13:35:06+00:00","remoteAddr":"","user":"--","app":"mail","method":"","url":"--","message":"not enough messages to train a classifier","userAgent":"--","version":"18.0.4.2"}

```
</details>
[image]

ovpc, 2020-08-25:
Looks like you are in the wrong place to me. 
"NextCloudPi" is not "NextCloudApp External Storage DropBox"

Grizaster, 2020-08-25:
OK I close it sorry!

closed 2020-08-25 by Grizaster