nextcloud/nextcloudpi #1167: Make Docker container immutable

gustafssone, 2020-08-26:
**Description**  
The docker container hosted at hub.docker.com is 5 months old. According to the documentation at https://docs.nextcloudpi.com/en/how-to-get-started-with-ncp-docker/ one should run "docker exec -it nextcloudpi ncp-update" and update, however if a container is redeployed, this information gets lost. This means for instance a "swarm deploy" command or "update" will reset to the version of ncp that currently exists at the Docker hub. Note that the actual nextcloud update (not ncp) persists after deploy/update. 
This is not a container best practice, it would be better to create a new build of the container so that the container is truly immutable, see https://cloud.google.com/solutions/best-practices-for-operating-containers#immutability or https://developers.redhat.com/blog/2016/02/24/10-things-to-avoid-in-docker-containers/ .

**Example**  
Set up an automated build https://docs.docker.com/docker-hub/builds/ . It should be possible to automate the build of new nextcloudpi versions whenever they are released as independent versions.

Keep up the good work!

ovpc, 2020-08-26:
We have no manpower to implement this, unless you are going to yourself. 
"Swarm deploy" would be way beyond the scope of NCP imho. 
But lets see what Nacho has to say. Thanks anyway.

gustafssone, 2020-08-26:
Thanks for the quick feedback! Well there are some input about swarm deployment here: https://ownyourbits.com/2019/08/21/make-your-nextcloudpi-highly-available/ . I use glusterfs and docker swarm myself, works great, but every time it is redeployed the version gets reset. Maybe it is possible to manually generate a new docker image whenever a new version of nextcloudpi is out? Not sure about the effort to implement this or how much extra work it is, but it should be to just run the Dockerfile and push it to the public repository.

The current setup of the docker image is not immutable, which docker containers should be :)

nachoparker, 2020-08-31:
Hi,

Thanks for the suggestion. Does `swarm update` not keep the volume? I don't see how that would be useful.

Also, keep in mind that this is an unconventional docker image in the sense that we are re-using the SDcard/VM/debian build scripts and dockerizing them. For this reason everything is one fat container as opposed to different ones (httpd, database, nextcloud and so on), which is also not standard docker practice.

It is done this way in order to be able to re-use the same build process for as many platforms as possible.

In any case, as Oliver said, we have no manpower to generate and test images frequently.

Unless somebody is willing to develop and own it (meaning maintaining it for a long time), I don't foresee any structural changes in the docker container.

closed 2020-08-31 by nachoparker

gustafssone, 2020-08-31:
Well it looks like it is reset at least. This means it is not very useful to neither run unattended upgrades or upgrade nextcloudpi within the container, as it is reverted upon a redeploy/update. This is outside data directory, which is of course kept.