nextcloud/nextcloudpi #1171: `ncp-update-nc 19.0.2` from 18.0.8 fails with a CLI exception: ProgressBar::advance() must be of the type int, string given

llucax, 2020-08-30:
Trying to upgrade from 18.0.8 to 19.0.2 using `ncp-update-nc 19.0.2` fails with the following exception:

`TypeError: Argument 1 passed to Symfony\Component\Console\Helper\ProgressBar::advance() must be of the type int, string given, called in /var/www/nextcloud/core/Command/Upgrade.php on line 144 and defined in /var/www/nextcloud/3rdparty/symfony/console/Helper/ProgressBar.php:310`

<details><summary>Upgrade log</summary>
$ sudo ncp-update-nc 19.0.2
Current   Nextcloud version 18.0.8.2
Available Nextcloud version 19.0.2
Download Nextcloud 19.0.2...                                                                                                                                                   
Back up current instance...
check free space...
Maintenance mode enabled
backup database...
backup files...
backup /var/www//nextcloud-bkp_20200828_1598646316.tar.gz generated
Maintenance mode disabled
Install Nextcloud 19.0.2...
Restore apps...                                                                                                                                                                
 * apporder
 * audioplayer
 * audioplayer_editor
 * calendar
 * camerarawpreviews
 * checksum
 * contacts
 * deck
 * dicomviewer
 * event_update_notification
 * extract
 * files_automatedtagging
 * files_downloadactivity
 * files_markdown
 * files_mindmap
 * files_photospheres
 * files_retention
 * flowupload
 * groupfolders
 * keeweb
 * mail
 * maps
 * metadata
 * news
 * nextcloudpi
 * notes
 * occweb
 * ocr
 * phonetrack
 * previewgenerator
 * suspicious_login
 * tasks
 * twofactor_nextcloud_notification
 * twofactor_totp
 * unsplash
 * video_converter
Fix permissions...
Upgrade...
Nextcloud or one of the apps require upgrade - only a limited number of commands are available                                                                                 
You may use your browser or the occ upgrade command to do the upgrade
Set log level to debug
Turned on maintenance mode
Updating database schema
Updated database
Disabled incompatible app: dicomviewer
Disabled incompatible app: files_automatedtagging
Disabled incompatible app: files_downloadactivity
Disabled incompatible app: files_photospheres
Disabled incompatible app: files_retention
Disabled incompatible app: keeweb
Disabled incompatible app: nextcloudpi
Disabled incompatible app: occweb
[... 206 lines cut ...]
2020-08-26 10:50:22 0 [Note] InnoDB: Completed initialization of buffer pool
2020-08-26 10:50:22 0 [Note] InnoDB: If the mysqld execution user is authorized, page cleaner thread priority can be changed. See the man page of setpriority().               
2020-08-26 10:50:23 0 [Note] InnoDB: 128 out of 128 rollback segments are active.
2020-08-26 10:50:23 0 [Note] InnoDB: Creating shared tablespace for temporary tables
2020-08-26 10:50:23 0 [Note] InnoDB: Setting file './ibtmp1' size to 12 MB. Physically writing the file full; Please wait ...                                                  
2020-08-26 10:50:23 0 [Note] InnoDB: File './ibtmp1' size is now 12 MB.
2020-08-26 10:50:23 0 [Note] InnoDB: Waiting for purge to start
2020-08-26 10:50:23 0 [Note] InnoDB: 10.3.23 started; log sequence number 5236372079; transaction id 11567282                                                                  
2020-08-26 10:50:23 0 [Note] InnoDB: Loading buffer pool(s) from /media/usb5tb-wdy8/nextcloud/mariadb/ib_buffer_pool                                                           
2020-08-26 10:50:23 0 [Note] Plugin 'FEEDBACK' is disabled.
2020-08-26 10:50:23 0 [Note] Server socket created on IP: '127.0.0.1'.
2020-08-26 10:50:24 0 [Note] Reading of all Master_info entries succeeded
2020-08-26 10:50:24 0 [Note] Added new Master_info '' to hash table
2020-08-26 10:50:24 0 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.3.23-MariaDB-0+deb10u1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Raspbian 10                                                                               
2020-08-26 10:50:50 0 [Note] InnoDB: Buffer pool(s) load completed at 200826 10:50:50

```
</details>

<details>
<summary>Nextcloud logs</summary>

Sorry, I'm not including this because they are just normal logs with access to my files, there are no errors or anything funny and I don't want to publish details about my personal files.
</details>

ovpc, 2020-08-30:
you are using  v1.28.1
try updating ncp to latest v1.29.0 before running nc-update-nc again
Also  disable app(s) that seems to be cause of trouble, re-enable after update complete.

llucax, 2020-08-30:
I see v1.29.0 was released very recently, I did an `ncp-update` before running `ncp-update-nc` but didn't find any updates at the time I did it. I just upgraded to v1.29.0 and the NC update went well. Thanks!

closed 2020-08-30 by llucax

feutl, 2020-08-31:
I do have the same problem, already on v1.29.8 of NCP and the update process on the WebUI (also tried it on the CLI) throw the exact same error.

Right now I am restoring the database via the roll-back feature, and I do hope this is going to work out fine again.
Any clues why this happened ? I am not sure if this is something NCP could check upfront

ovpc, 2020-08-31:
> I do have the same problem, already on v1.29.8 of NCP and the update process on the WebUI (also tried it on the CLI) throw the exact same error.
>
> Right now I am restoring the database via the [...]

Please use the forum for help. Run ncp-report and share output via pastbin or similar.

feutl, 2020-08-31:
will do thanks
https://help.nextcloud.com/t/an-unhandled-exception-has-been-thrown-when-upgrading-to-19-0-2-with-ncp/91020

nachoparker, 2020-09-01:
Nextcloud bug, seems like triggered by the OCR app

https://github.com/nextcloud/server/issues/21067

feutl, 2020-09-01:
@ovpc and @nachoparker good to know - great and thanks for any help and advice 👍