nextcloud/nextcloudpi #1177: [Feedback] Updating to version 20.*

sunjam, 2020-09-04:
```
NextCloudPi version	v1.25.0
NextCloudPi image	NextCloudPi_docker_04-06-20
Nextcloud check	ok
Nextcloud version	19.0.2.2
Beta Version            20.0.0beta3
```

From 19.0.2.2 stable build on Docker arm64.  Manually updated from 18.0.3 to 19.0.2 via stable release of ncp-config.

[Output: Updating from 19.0.2.2 to 20.0.0beta3 via occ on command line](https://gist.github.com/sunjam/808a89e201bb83b228c46f29cbca9e8e)

If updating via ncp-web via `20.0.0beta3` the following error is seen:

```
[ nc-update-nextcloud ] (Fri Sep 4 17:56:51 UTC 2020)
Cannot load Zend OPcache - it was already loaded
Current Nextcloud version 19.0.2.2
Available Nextcloud version 20.0.0beta3
Download Nextcloud 20.0.0beta3...
Error downloading
Clean up...
```

Nextcloud system update from `Stable` to `Beta` within webui also fails.

```
Preparing update
Set log level to debug
An error occurred.
```

Occ command upgrade succeeds, but only when occ path is fully defined:

`sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --on`
`sudo -u www-data php /var/www/nextcloud/occ upgrade`
`sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --off`

`20.0.0beta3` will run successfully. :+1:

sunjam, 2020-09-05:
NextcloudPi app works perfectly.  **Just needs to be updated to version 20 for usual installation warning from the appstore.**

sunjam, 2020-09-27:
Auto-update of ncp is on, but Nextcloud itself has not been able to update from 20.0.0 beta4 to rc1 (or now rc2).  Seems it might be related to the numbering scheme of the release clients vs the beta.  I've been applying each update maually, with no luck for nc-nextcloud interface.

Update current instance to a new Nextcloud version
Version: 2.0.0.0RC2

**Output from nc-update**
[ nc-update-nextcloud ] (Sun Sep 27 15:28:28 UTC 2020)
Current Nextcloud version 20.0.0.6
Available Nextcloud version 20.0.0RC2
Nothing to update

**What is displayed in Nextcloud / Admin / Settings**
NextCloudPi 20.0.0 Beta 4

A new version is available: Nextcloud 20.0.0 RC2
5 apps have no update for this version available

**Open Updater** fails to actually do anything, then suddenly it worked directly from Nextcloud Adming Settings page and fails at this step:

```
Create backup
Could not copy "/data/nextcloud/apps/dashboard/js/dashboard.js.map" to "/data/nextcloud/data/updater-ochxgtum9iy3/backups/nextcloud-20.0.0.6/apps/dashboard/js/dashboard.js.map"

```

Saroufim, 2020-10-09:
The upgrade works mostly fine but I'm facing issues with matterbridge. Nextcloud asks me to install it manually and when I do it complains about not being able to execute the binary. I tried @sunjam's workaround but it didn't work. I keep hitting the same issue.

sunjam, 2020-11-05:
Updated to 20.0.1 without issue on Nextcloud Docker arm64 from 20.0.0
- Matterbridge fails to install from Settings/Talk menu as mentioned above.
- Collabora arm64 server installs, but fails to open on 2gb ram or less.

sirdrakeistda, 2020-11-17:
I upgraded to 20.0.1 too. But I have a few issues.
```NextCloudPi version	v1.31.0
Nextcloud check	ok
Nextcloud version	20.0.1
php Version             7.4.12
```

The Nextcloud Web Interface runs without any Problems, but connecting with an App (it makes no difference which App, e.g. DavDroid, News, Notes, or the desktop sync client,...) have very often connection Problems because of connection timeouts. In my Nextcloud Logs I see different errors but they don't seem to be related to the App connection Problems. (Maybe they will be fixed with NC20.0.2)
Matterbridge does not work for me. Neither one of the two Web installer (one in the settings and one for all apps) or the occ installer or the installation way mentioned earlier works for me correctly. Matterbridge gets installed, but does not work (In the settings the installation of matterbridge is not recognized)
The Update to php 7.4 worked without any problems.

feutl, 2020-11-20:
If the upgrade is getting automated via NCP please take this in consideration, otherwise the upgrade will fail
https://github.com/nextcloud/mail/issues/3784#issuecomment-707732762

nachoparker, 2020-11-25:
@feutl did you run into this?

https://github.com/nextcloud/news/issues/820

That's what I am running into

nachoparker, 2020-11-25:
follow up -> https://github.com/nextcloud/news/issues/820#issuecomment-733476369

nachoparker, 2020-11-25:
I had to add ncc db:add-missing-primary-keys to the updater. Other than that, the news issue is the only thing preventing me from publishing this upgrade

nachoparker, 2020-11-25:
I had to add `ncc db:add-missing-primary-keys` to the updater. Other than that, the news issue is the only thing preventing me from publishing this upgrade

feutl, 2020-11-25:
@nachoparker 
Yes I ran into both issues. Sorry for not reporting back my experience, I forgot to document it well enough. I started over using the NCP updater, which failed and rolled back on my productive environment.
Afterwards I used the WebUpdater, which run through until the NEWS app failed to get activated, I was stuck in maintenance mode. Anyhow, used the CLI updater to understand the error and it was in fact the news app. I deactivated the news app, ran the CLI updater again (not NCP), which worked (as far as I can remember).
Afterwards I logged into the system, updated all missing apps and activated the news app (as well as others like calender, talk and some I cannot remember) manually. All seems to work fine, last piece missing were the primary keys.

I must say, I more often used the built in updater of NC instead of the NCP updater to upgrade major versions, the rollback of NCP with my installation takes forever, if something fails, and most of the time something fails. Especially if you be one of the first ;) The NC updater allows you to fix problems before a rollback is initiated automatically and works quite well to be honest. Or said differently, worked well the last few times from 18 to 19 and 19 to 20 :)

Perhaps it is time to reconsider the NC Updater, not ditching it, but make the automatic rollback "optional" ;) so someone fixes the issue(s) which most are easy fixable - or decides to rollback if this person does not want to invest time.

Hope this helped somehow, if I find some of my notes, spread somewhere I add them to the list.

nachoparker, 2020-11-25:
No worries, everything is fixed now.

There is a reason why we have auto-updates and nextcloud updates with version "0", which means latest tested update.

Of course if you want to try the latest before I do, you'll encounter the issues that I fix so others don't run into them :)

If that's what you want to do, you might as well just use the NC official updater.

Cheers

closed 2020-11-25 by nachoparker

feutl, 2020-11-25:
@nachoparker 
At the end, all of this is the reason why I still stick with NCP, especially every time an updated failed ;) I knew and "tested" the rollback again - and it worked. In 99% this is how it should be 👍 

Thanks again for the great work with NCP, without NCP I would not have migrated my family and some friends to Nextcloud :) and we do store a few thousand calendar and contact entries in NC, and sync over 2TB of data with it. And the younger family members have a way to communicate with NC Talk in a secure way in this special times :)

THANKS a thousand times

nachoparker, 2020-11-25:
our pleasure :)

sirdrakeistda, 2020-12-14:
I fixed the matterbridge issue. If you run Nextcloud on a Raspberry Pi (only possible for Raspberry Pi 3 and higher), first upgrade to a 64-bit system. This could be done by ```rpi-update``` (not sure if necessary) and then editing ```/boot/config.txt``` and edit ``` arm_64bit=1```
Reboot your system and  then run the matterbridge 64 bit binary with:
``` sudo -u www-data php occ config:app:set spreed matterbridge_binary --value="/var/www/nextcloud/apps/talk_matterbridge/bin/matterbridge-1.20.0-linux-arm64" ```

NekoTherion, 2021-10-20:
could use a BSD binary for those of us on TrueNAS plugin ;) i had to install from pkg, and point it to the system-wide install with occ