nextcloud/nextcloudpi #1180: NCP Preview

daschmidt1994, 2020-09-09:
The ncp preview generatore don't work anymore.
The output from the log is:

In Application.php line 598:
                                                                    
  [Symfony\Component\Console\Exception\NamespaceNotFoundException]  
  There are no commands defined in the "preview" namespace.         
                                                                    

Exception trace:
  at /data/nextcloud/3rdparty/symfony/console/Application.php:598
 Symfony\Component\Console\Application->findNamespace() at /data/nextcloud/3rdparty/symfony/console/Application.php:651
 Symfony\Component\Console\Application->find() at /data/nextcloud/3rdparty/symfony/console/Application.php:236
 Symfony\Component\Console\Application->doRun() at /data/nextcloud/3rdparty/symfony/console/Application.php:148
 Symfony\Component\Console\Application->run() at /data/nextcloud/lib/private/Console/Application.php:215
 OC\Console\Application->run() at /data/nextcloud/console.php:100
 require_once() at /data/nextcloud/occ:11

NextCloudPi version  v1.29.8
NextCloudPi image    NextCloudPi_docker_04-05-20
distribution         Debian GNU/Linux 10 \n \l
automount            no
USB devices          synoboot 
datadir              /data/nextcloud/data
data in SD           yes
data filesystem      ecryptfs
data disk usage      1.3T/1.8T
rootfs usage         1.3T/1.8T
swapfile             /dev/zram3
dbdir                /data/database
Nextcloud check      ok
Nextcloud version    19.0.2.2
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   ***REMOVED SENSITIVE VALUE***
gateway              ***REMOVED SENSITIVE VALUE***
interface            eth0
certificates         ***REMOVED SENSITIVE VALUE***
NAT loopback         no
uptime               5days

nachoparker, 2020-09-12:
also tested this and works on my machine

closed 2020-09-12 by nachoparker

tbruckmaier, 2020-09-26:
I ran into the same issue, would get the same error every night per mail from cron. I found this issue https://github.com/nextcloud/nextcloud-snap/issues/905 which said to enable the preview generator app. While there was a warning that the app was "untested with my current nextcloud version 19.0.3", I enabled it and the error is gone.

Maybe the script `nc-previews` should check if that app is enabled.