nextcloud/nextcloudpi #1182: Libravatar federation as client

tclavier, 2020-09-10:
**Description**  
As a user of nextcloud server, i want to see my Libravatar associate with my email adresse in my account. 

**Example**  
As user `thomas` on server `nextcloud.domain.com` with email ̀`thomas@domain.com` I whant to see in my account the corresponding photo to my libravatar photo. To do that, my nextcloud server follow the libravatar api (https://wiki.libravatar.org/api/) 

* search in DNS the entry : `SRV _avatars-sec._tcp._tcp.domain.com`, no response we must fallback to default cdn
* compute the md5 or sha256 of my lower case email (`thomas@domain.com`): 4fbeb1bd99d764f65642babb1308c9c9
* request the compute URL : https://cdn.libravatar.org/avatar/4fbeb1bd99d764f65642babb1308c9c9
* finaly my nextcloud return the downloaded image.

mrschaal, 2021-09-04:
Would be a great enhancement for Nextcloud server, especially in corporate environments!

ovpc, 2021-09-04:
I think this is not  NCP related.
Maybe try (feature request) at Nexctloud (mail) app thread.
Unless someone does a PR, I'm closing this for now.

tclavier, 2021-09-04:
Please, reopen this ticket, it's not for email app, it's to use a libreavatar to generate https://nextcloud.domain.com/avatar/user/64

ovpc, 2021-09-04:
@tclavier  Still think it is not for ncp but for nc devs.
Reopened, are you up for a PR yourself?

tclavier, 2021-09-04:
hum .... yes, with help.