nextcloud/nextcloudpi #1187: ncp-update 1.28.1 failed

jacremer, 2020-09-18:
Hi,

ncp is running fine on a RPI3+ but fails to update 1.28.1.

NextCloudPi v1.28.1 is outdated
update to v1.29.8 through 'ncp-config' or type 'sudo ncp-update'

pi@nextcloudpi:~ $ sudo ncp-update
Downloading updates
Performing updates
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done

exfat-fuse is already the newest version (1.3.0-1).
exfat-utils is already the newest version (1.3.0-1).
file is already the newest version (1:5.35-4+deb10u1).
haveged is already the newest version (1.9.1-7).
lsb-release is already the newest version (10.2019051400+rpi1).
php-smbclient is already the newest version (0.9.0-3+b1).
php7.3-gmp is already the newest version (7.3.19-1~deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 176 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up dphys-swapfile (20100506-5+rpt1) ...
Job for dphys-swapfile.service failed because the control process exited with error code.
See "systemctl status dphys-swapfile.service" and "journalctl -xe" for details.
invoke-rc.d: initscript dphys-swapfile, action "restart" failed.
 dphys-swapfile.service - dphys-swapfile - set up, mount/unmount, and delete a swap file
   Loaded: loaded (/lib/systemd/system/dphys-swapfile.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/dphys-swapfile.service.d
           ncp-delay-automount.conf
   Active: failed (Result: exit-code) since Fri 2020-09-18 10:01:35 CEST; 43ms ago
     Docs: man:dphys-swapfile(8)
  Process: 11650 ExecStartPre=/bin/sleep 30 (code=exited, status=0/SUCCESS)
  Process: 11658 ExecStart=/sbin/dphys-swapfile setup (code=exited, status=0/SUCCESS)
  Process: 11681 ExecStart=/sbin/dphys-swapfile swapon (code=exited, status=255/EXCEPTION)
 Main PID: 11681 (code=exited, status=255/EXCEPTION)

Sep 18 10:01:05 nextcloudpi systemd[1]: Starting dphys-swapfile - set up, mount/unmount, and delete a swap file...
Sep 18 10:01:35 nextcloudpi dphys-swapfile[11658]: want /media/USBdrive/swap=2048MByte, checking existing: keeping it
Sep 18 10:01:35 nextcloudpi dphys-swapfile[11681]: swapon: /media/c9af4f9f-f6c6-4304-b1b9-38c32ac2569a/swap: insecure permissions 0775, 0600 suggested.
Sep 18 10:01:35 nextcloudpi dphys-swapfile[11681]: swapon: /media/c9af4f9f-f6c6-4304-b1b9-38c32ac2569a/swap: swapon failed: Device or resource busy
Sep 18 10:01:35 nextcloudpi systemd[1]: dphys-swapfile.service: Main process exited, code=exited, status=255/EXCEPTION
Sep 18 10:01:35 nextcloudpi systemd[1]: dphys-swapfile.service: Failed with result 'exit-code'.
Sep 18 10:01:35 nextcloudpi systemd[1]: Failed to start dphys-swapfile - set up, mount/unmount, and delete a swap file.
dpkg: error processing package dphys-swapfile (--configure):
 installed dphys-swapfile package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 dphys-swapfile
E: Sub-process /usr/bin/dpkg returned an error code (1)

**System information**

<details>
<summary>NextCloudPi diagnostics</summary>

```

Cannot load Zend OPcache - it was already loaded
NextCloudPi version  v1.28.1
NextCloudPi image    NextCloudPi_02-06-18
distribution         Raspbian GNU/Linux 10 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
[... 146 lines cut ...]
```

{"reqId":"WAaZnXPDkZNPOJpVNmKH","level":2,"time":"2020-09-18T04:30:04+00:00","remoteAddr":"","user":"--","app":"fulltextsearch","method":"","url":"--","message":"Exception while cronIndex: OCA\\FullTextSearch\\Exceptions\\PlatformNotSelectedException - Admin have not selected any IFullTextSearchPlatform","userAgent":"--","version":"19.0.2.2"}
{"reqId":"ORSOYGR93MM5iNvjJifK","level":2,"time":"2020-09-18T04:45:03+00:00","remoteAddr":"","user":"--","app":"fulltextsearch","method":"","url":"--","message":"Exception while cronIndex: OCA\\FullTextSearch\\Exceptions\\PlatformNotSelectedException - Admin have not selected any IFullTextSearchPlatform","userAgent":"--","version":"19.0.2.2"}
{"reqId":"fBHgVTj4qYYb1nPsdmU8","level":2,"time":"2020-09-18T05:00:02+00:00","remoteAddr":"","user":"--","app":"fulltextsearch","method":"","url":"--","message":"Exception while cronIndex: OCA\\FullTextSearch\\Exceptions\\PlatformNotSelectedException - Admin have not selected any IFullTextSearchPlatform","userAgent":"--","version":"19.0.2.2"}
{"reqId":"IGETmZtKfZ7wOrXtfPVb","level":2,"time":"2020-09-18T05:15:02+00:00","remoteAddr":"","user":"--","app":"fulltextsearch","method":"","url":"--","message":"Exception while cronIndex: OCA\\FullTextSearch\\Exceptions\\PlatformNotSelectedException - Admin have not selected any IFullTextSearchPlatform","userAgent":"--","version":"19.0.2.2"}
{"reqId":"8K1axqiDX28VChmo0Wx0","level":2,"time":"2020-09-18T05:30:04+00:00","remoteAddr":"","user":"--","app":"fulltextsearch","method":"","url":"--","message":"Exception while cronIndex: OCA\\FullTextSearch\\Exceptions\\PlatformNotSelectedException - Admin have not selected any IFullTextSearchPlatform","userAgent":"--","version":"19.0.2.2"}
{"reqId":"X2RIac0j0D8rs657dxmQfwAAAAA","level":3,"time":"2020-09-18T05:40:57+00:00","remoteAddr":"192.168.2.51","user":"--","app":"index","method":"GET","url":"/index.php/204","message":{"Exception":"Doctrine\\DBAL\\DBALException","Message":"Failed to connect to the database: An exception occurred in driver: SQLSTATE[HY000] [2002] No such file or directory","Code":0,"Trace":[{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php","line":889,"function":"connect","class":"OC\\DB\\Connection","type":"->","args":[]},{"file":"/var/www/nextcloud/lib/private/DB/Connection.php","line":194,"function":"executeQuery","class":"Doctrine\\DBAL\\Connection","type":"->","args":["SELECT * FROM `oc_appconfig`",[],[],null]},{"file":"/var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Query/QueryBuilder.php","line":200,"function":"executeQuery","class":"OC\\DB\\Connection","type":"->","args":["SELECT * FROM `oc_appconfig`",[],[]]},{"file":"/var/www/nextcloud/lib/private/DB/QueryBuilder/QueryBuilder.php","line":216,"function":"execute","class":"Doctrine\\DBAL\\Query\\QueryBuilder","type":"->","args":[]},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":331,"function":"execute","class":"OC\\DB\\QueryBuilder\\QueryBuilder","type":"->","args":[]},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":109,"function":"loadConfigValues","class":"OC\\AppConfig","type":"->","args":[]},{"file":"/var/www/nextcloud/lib/private/AppConfig.php","line":287,"function":"getApps","class":"OC\\AppConfig","type":"->","args":[]},{"file":"/var/www/nextcloud/lib/private/legacy/OC_App.php","line":883,"function":"getValues","class":"OC\\AppConfig","type":"->","args":[false,"installed_version"]},{"file":"/var/www/nextcloud/lib/private/Server.php","line":663,"function":"getAppVersions","class":"OC_App","type":"::","args":[]},{"file":"/var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/ServerContainer.php","line":124,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"/var/www/nextcloud/lib/private/Server.php","line":1701,"function":"query","class":"OC\\ServerContainer","type":"->","args":["OC\\Memcache\\Factory"]},{"file":"/var/www/nextcloud/lib/private/Server.php","line":854,"function":"getMemCacheFactory","class":"OC\\Server","type":"->","args":[]},{"file":"/var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":118,"function":"OC\\{closure}","class":"OC\\Server","type":"->","args":["*** sensitive parameters replaced ***"]},{"file":"/var/www/nextcloud/lib/private/ServerContainer.php","line":124,"function":"offsetGet","class":"Pimple\\Container","type":"->","args":["OC\\App\\AppManager"]},{"file":"/var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php","line":163,"function":"query","class":"OC\\ServerContainer","type":"->","args":["OC\\App\\AppManager"]},{"file":"/var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php","line":114,"function":"OC\\AppFramework\\Utili [... cut ...] rch\\Exceptions\\PlatformNotSelectedException - Admin have not selected any IFullTextSearchPlatform","userAgent":"--","version":"19.0.2.2"}
{"reqId":"GrsFVaswO2pGns9qT2l2","level":2,"time":"2020-09-18T08:00:03+00:00","remoteAddr":"","user":"--","app":"fulltextsearch","method":"","url":"--","message":"Exception while cronIndex: OCA\\FullTextSearch\\Exceptions\\PlatformNotSelectedException - Admin have not selected any IFullTextSearchPlatform","userAgent":"--","version":"19.0.2.2"}

```
</details>

ovpc, 2020-09-18:
Looks like copy of https://github.com/nextcloud/nextcloudpi/issues/713
and
https://help.nextcloud.com/t/ncp-curl-installer-on-lxc-swap-errors/91986

If you had an issue with dphys-swap, updating ncp is not going to be resolve it.

jacremer, 2020-09-18:
Thanks Oliver. I already saw the messages and i tried dpkg --configure -a and got the same error. It's a long time running application not a fresh one. Since early September i get notifications regarding the (automatic) installation of 1.28.1 so i uses shh for manual interaction with the installation (last one is more than one year ago, i hardly remembered the password :) ). I noticed two things in the reporting:
1. ... 1 not fully installed or removed ... i  have no clue which file is meant.
2. ... swapon failed: Device or resource busy ... but no errors in Nextcloud configuration.

sudo swapon -s returns:
FilenameTypeSizeUsedPriority
/dev/zram0                             partition2497002565
/dev/zram1                             partition24970005
/dev/zram2                             partition24970005
/dev/zram3                             partition24970005
/media/c9af4f9f-f6c6-4304-b1b9-38c32ac2569a/swapfile    20971480-2

closed 2020-11-27 by nachoparker

jacremer, 2020-11-27:
Hi nachoparker,You closed the issue but it isn't solved yet. The referenced issue all have to do with new installations. This is a already long time running application for which this update is failing. On the forums nothing can be found that's alike? You have a clue why this update is failing and what I can do about it? Thanks a lot.Regards, Jac

Closed #1187.

LSchober, 2021-02-23:
Hi @nachoparker, hi @jacremer,
i've got the same issue...my installation is also a very long time running one.
I'm also stuck at ncp 1.28.1. Is there a solution yet? What can I do about it?
Best regards