nextcloud/nextcloudpi #1188: High Memory and CPU Usage with MySQL

feutl, 2020-09-24:
I am running into performance issues with my NCP installation.
Specifically said `mysqld` eats up lots of ram and cpu as a consequence of swapping - I must assume.

I looked up the NCP history and found
https://github.com/nextcloud/nextcloudpi/commit/06017a4bcf8d2ee868716cfa7b7bc65a06bad58d

I wonder why `query_cache` is set to 0, which I think was not the case prior, also I try to understand how this calculation is working, in my case I have around 40 mysqld threads, all of them have 2119M reserved and 4530M virtual memory allocated - `htop` says.

I do have a 8GB machine running, and the nc-limits are

-  maximum file size: 10G
- memory limit: 0
- php threads: 40
- redis memory: 0

If someone asks why I am not on the nextcloud support forum, because I highly think this could be a bug, or some automated miscalculation for non-PI systems.

feutl, 2020-09-24:
I add a discussion from the NC Forum
https://help.nextcloud.com/t/high-mysql-cpu-usage/91084/2

I do not know yet if this person uses NCP too.

feutl, 2020-09-24:
Another update from my site, I brought down the `mysqld` memory and cpu usage, especially the cpu usage by turning on `query_cache = 1` again.

Honestly I have no idea when this was turned off, or if it always was off and something else changed, but I felt a slow responding NC instance and I must assume that this happened when moving to NCP supporting NC19 and upgrading my instance to NC19 on that road.

nachoparker, 2021-10-18:
Query cache is discouraged and it has been straight up removed in recent versions

https://web.archive.org/web/20210302175500/https://mysqlserverteam.com/mysql-8-0-retiring-support-for-the-query-cache/

I am going to run some tests nonetheless to try to optimize mysql again, I'll post here soon.