nextcloud/nextcloudpi #1192: Add options to install and enable missing packages from apt

sunjam, 2020-10-10:
#1125, option to add `smbclient` from the ncp menus.
#1177 add option to add `php-imageck` and `bmath` from the ncp menus.

This feature request is for installing packages / dependencies missing within the default NextcloudPi installations.  The current method is for the admin to install these via apt-get, but that is not persistent in Docker method.  I understand these are not included by default, but it would make sense to make them a one click fix, which is also in-line with the way Docker images are designed.

Thanks for considering!

sunjam, 2020-10-11:
If considered, [method that would make the most sense for non-interactive is](https://pythonspeed.com/articles/system-packages-docker/):

$ apt-get update
$ apt-get -y install --no-install-recommends $PACKAGE
$ apt-get clean

This avoid installing unnecessary additions outside the required packages in apt-get.

segator, 2021-03-06:
The problem to do it by hand is that if you reinstall the docker everything is lost.. something that in case we use kubernetes to deploy applications is a very common case :(
we need a way to inject an script in runtime to install packages.

C-Fu, 2021-06-06:
I think this would be awesome and probably needed for CUPS and the Printer App https://github.com/nextcloud/nextcloudpi/issues/1288

nachoparker, 2021-10-15:
I think that the best way of achieving this is to build a new docker image with the NCP image as a base, and install any extra stuff you might want

closed 2021-10-15 by nachoparker