nextcloud/nextcloudpi #1195: Restore(/Backup) Strategy for Large Amount of Data

paschaef, 2020-10-13:
Recently, I had to restore my nextcloud instance with the `nc-restore` functionallity of nextcloudpi. When using `nc-restore` a temporary directory is created (`mktemp` [here](https://github.com/nextcloud/nextcloudpi/blob/9d65011e84f46037d99f2ceea765b4be3d1031ae/bin/ncp/BACKUPS/nc-restore.sh#L33)) where the backup tar-file is extracted into. Aftwards, the extracted data is moved from `TMPDIR` to the `DATADIR`.

However, with this approach the double amount of disk space is required during the restoration process (once disk space for the tar-file and once for the extracted files). This is especially severe if there is a large amount of data in the data directory - in my case it was 1.5TB which required me to have 3TB of disk space during the restoration process. I think it is not always possible for everyone to provide double amount of disk space only to restore the backup data.

I would like to open a discussion for a new strategy for the restoration process, so that there is no data duplication (tar-file + extracted files). Maybe an approach would be to add an additional option to provide a `DATADIR_Restore` path, where the "ready-to-move" data is located. The content of this `DATADIR_Restore` directory should be the same as the `TMPDIR` after the [extract command](https://github.com/nextcloud/nextcloudpi/blob/9d65011e84f46037d99f2ceea765b4be3d1031ae/bin/ncp/BACKUPS/nc-restore.sh#L55) in [nc-restore.sh](https://github.com/nextcloud/nextcloudpi/blob/9d65011e84f46037d99f2ceea765b4be3d1031ae/bin/ncp/BACKUPS/nc-restore.sh). Maybe, one also has to adapt the [nc-backup.sh](https://github.com/nextcloud/nextcloudpi/blob/c63cb274e9a831ef0fb54d13575db3b29e84d8cb/bin/ncp/BACKUPS/nc-backup.sh), i.e. to backup the data into a directory instead of a tar-file.

I'm just sharing my thoughts and ideas about this topic, not claiming that this is the perfect solution. What do you think about it? Any ideas on how not to double the data during the restoration process? 
I think it is worth to discuss an appropriate strategy, before starting to implement it and making a pull request.

---
Side note: I use [nc-backup.sh](https://github.com/nextcloud/nextcloudpi/blob/c63cb274e9a831ef0fb54d13575db3b29e84d8cb/bin/ncp/BACKUPS/nc-backup.sh) to backup only the database and do the backup of the data directory manually. To restore, I first append my data directory backup to the tar-file generated by nc-backup.sh (mimicking the behaviour of nc-backup.sh when include the data dir), then I use [nc-restore.sh](https://github.com/nextcloud/nextcloudpi/blob/9d65011e84f46037d99f2ceea765b4be3d1031ae/bin/ncp/BACKUPS/nc-restore.sh) to restore the backup tar-file.

AN-NCuser459, 2020-10-13:
I've run into the same problem trying to restore a 1.1TB database using a 2TB drive (and i'm frustrated). If there were a simple option to make an UNCOMPRESSED back-up, and restore an UNCOMPRESSED back-up I think this would alleviate this problem, and i think it would be very simple to code. Alternatively, being able to specify the location of the temporary file would also be an option allowing an additional HD to be used for that (for example).
Sorry if i missed something, i'm new to NC and coding.

paschaef, 2020-10-14:
@nachoparker what do you think of an additional option to backup/restore to/from a directory instead of a tar-file? The contents of the directory and the tar-file are idendical (database and data dir) if compression is disabled.

ovpc, 2020-10-14:
@paschaef 
nc-backup already has an option to compress : yes or no.
[image]

I have auto backups enabled without data, so it never gets to big. For data I use nc-snapshots

paschaef, 2020-10-14:
@ovpc the issue is not the compression. As explained above, the issue is that the backup is always stored in a tar-file whether compression is enabled or not. When you restore this backup you first have to extract all files from the tar-file (again doesn't matter if compression is enabled or not), which needs the same amount of disk space for the extracted data as the tar-file itself, resulting in needing twice as much disk space.

I also have auto backup enabled without data, for the data I use `borg` external tool to backup. However, at the moment there is no option to restore ncp with a backup-file (tar-file) generated by `auto backup` without data, and providing the data itself in another directory. Therefor, I manually add (as described above) my data backup to the tar-file, which has the same effect as I would have backup up ncp **with** data.

Therefore, I think we should provide an option to restore (and backup) database and data in a non tar-file manner.

theCalcaholic, 2020-10-19:
A solution could be to mount - instead of extract the archives... :thinking: 

Should be possible somehow (e.g. via `archivemount`)

paschaef, 2020-10-19:
This approach sounds very interesting.
This could potentially solve the problem.

I'm just thinking if it would be profitable to provide a restore mechanism that allows for a separate data directory, if the user backs up the data directory manually, this can be in addition to the approach with `archivemount`.

theCalcaholic, 2020-10-19:
Scratch that, I gave it some more thought. :D
It would probably be easier (and less error prone) to just extract the archive directly to the correct partition.

In regards to manual backups - if you're able to create manual backups, you should also be able to manually restore them, imo.
An automated mechanism for that purpose would be very insecure, because it would make a lot of assumptions about how the backup was created.

paschaef, 2020-10-20:
Yes, I think you're right. The user should provide the already extracted backupfile to a directory from where NCP can restore it.

I agree, regrading the manual backups, the user then should take care of backup/restore by himself.

basxto, 2020-11-21:
Doesn’t it work to put nextcloud into maintenance mode, restore the data and then restore the backup without data?
That was what I was aiming for.

closed 2021-10-15 by nachoparker