nextcloud/nextcloudpi #1200: One SHOULD **NOT** have to type in a huge long massive multi-random-character password at activation.

madscientist42, 2020-10-26:
**Description**  

Security is **_laudable_**, but this is beyond **_INSANE_**.   You want people to actually, you know, **_USE_** your project and the average user will be so put-off by this that they will not use it.  Your userbase isn't developers or security researchers- it's the average user which **_WILL_** walk away from this at this point.  If you want security, I suggest **_PASSPHRASES_**, random sequences of human intelligible, but not easily zoomed by a computer, sequences of english words.  Something that an end user will gripe about, but will be willing to DO.  

What you have is **_**BROKEN**_** by design.

ovpc, 2020-10-26:
Most of us are fine with using copy/paste and storing long and safe passwords in password managers. I use the NC passman app, and used lastpass before that. I'd suggest you start using them too.
Security is not just laudable, one cannot do without it these days. In fact a good password is not enough, that is why 2FA is used to take security up a notch.

madscientist42, 2020-10-26:
Simply put: https://xkcd.com/936/

You need to learn the distinction there and this is embarrassing.  How is
copypasting an impossible to type password (yes) secure?  And you have
someone print it out.

YOU aren't the customer, you're the dev.  Customers are your mother or that
little grandmother that sort-of does tech.  It will never get serious
traction with it that way.

closed 2020-10-26 by ovpc

madscientist42, 2020-10-27:
Use the forums for this?  This is a feature request.  It's telling that you did this the way you did.  It actually doesn't work that way, you know.  You have the same mentality of the Pwnie Award Winning systemd, with all that this entails.

ovpc, 2020-10-27:
If you payed attention, you'd notice I removed the "use the forum please ..." tag
It was what I initially thought, but after re-reading, changed it to "question"
I've added the "feature request" if it pleases you.
How many users would agree with your point of view, do you think? 
Feel free to use the [forum](https://help.nextcloud.com/tag/ncp) to find supporters for your quest.

Thank you for wanting to contribute.