nextcloud/nextcloudpi #1201: Upload - Unknown Error after updating to 19.0.4

m-gorn, 2020-10-27:
Nextcloud version (eg, 18.0.2): 19.0.4
Operating system and version (eg, Ubuntu 20.04): nextcloudpi
Operating system: Linux 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l
Webserver: Apache (fpm-fcgi)
Database: mysql 10.3.22
PHP version: 7.3.19-1~deb10u1

After updating nextcloud to 19.0.4 i can’t upload files bigger than 10KB. The Error ist “unknown upload error” or “connection lost”.

|datadir|/media/myCloudDrive/ncdata|
|data in SD|no|
|data filesystem|btrfs|
|data disk usage|187G/3.7T|

[image]

janborg, 2020-10-31:
I struggled with the same issue for a while and could finally fix it. I cant say why or exactly since when, but the `tmp` folder set in the `/etc/php/7.3/php.ini` (`sys_temp_dir = /var/www/nextcloud/data/tmp`) didn't exist (anymore).

As my nextcloud data is on a USBDrive, the path to my ncdata-folder is `/media/myCloudDrive/ncdata/`. So I made a `/tmp` folder there and set the php.ini to this folder: `sys_temp_dir = /media/myCloudDrive/ncdata/tmp'.

After restarting php-fpm, I could upload again without any problem.

aaronsama, 2020-11-01:
Thanks a lot @janborg! I have been struggling with this for a few days now and was about to reinstall the whole thing. Unfortunately it seems that the recent update may have changed some configuration. This is how I fixed it:

1. open `/etc/php/7.3/fpm/php.ini`
2. edit the config value for `sys_temp_dir` to the `ncdata/tmp` directory already in my external USB drive
3. put the same value for `upload_tmp_dir` (in theory it should be enough to just comment this out according to the comments above it)
4. restart php-fpm with `sudo service php7.3-fpm restart`

Unfortunately I'm completely ignorant about PHP 😞 . If it may help, my instance has been going on without issues and regular (automated) updates since March last year. Maybe some breaking change was introduced?

jeanrjc, 2020-11-20:
I have the same issue. It appears that the data folder under `/var/www/nextcloud/` was present prior to the update to 19.0.4, and is not present anymore now. 

Also, I don't know whether it's related, but some folder was erased on one of my computer and on the server, and my other computer was trying to update the erased files but couldn't due to this error. I'm still figuring out how some folder could have been erased on my computer and on the server (there are not in the. Given that I updated nextcloud (which went fine supposedly) yesterday, it looks odd for a coincidence. 

Anyway, if someone could investigate on that, it would be great.

Thanks

alxtrnr, 2021-01-01:
Thanks for solving this. I got the

> Error: Cannot modify header information - headers already sent by (output started at /var/www/nextcloud/3rdparty/sabre/http/lib/Sapi.php:132) at /var/www/nextcloud/apps/dav/lib/Connector/Sabre/File.php#691

No uploads would work without this fix. With NC 20.0.3 / NCPi v1.33.0

This seems to have fixed it for now:

1. open `/etc/php/7.3/fpm/php.ini`
2. edit the config value for `upload_tmp_dir` to the `ncdata/tmp` directory already in my external USB drive
3. restart php-fpm with `sudo service php7.3-fpm restart`

closed 2021-01-01 by nachoparker