nextcloud/nextcloudpi #1202: Docs Proposal: Forum, Github or Telegram chat: which to use when ?

ovpc, 2020-10-29:
Proposal for new post at docs.nextcloudpi.com, There have been concerns about occasional spamming.  Idea is to clarify usage of the platforms we have available.

### Forum, Github or Telegram chat: which to use when ?

Three channels, they basically are for:
### Forum

    NC and NCP Support in general

### Github

    Bug reports
    feature requests
    docs related issues

### Telegram

    General discussion ncp and nc related
    Simple yes/no answerable questions. Such a question typically does not exceed one or two lines and is contained in a single message. Sending multiple single line messages will be considered rude and you risk getting banned for spamming.
    Posting links to github issues or forum posts.

###Telegram wiki team

Discuss documention. 
Join wiki team, request login to WP docs, to contribute, translate or edit

ovpc, 2020-10-30:
published https://docs.nextcloudpi.com/en/forum-github-or-telegram-chat-which-to-use-when/

sunjam, 2020-10-31:
*Minor editing for forum section*
```
[Nextcloud](https://help.nextcloud.com/c/support/7) and [NCP](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm/33) Support in general, use [ncp tag](https://help.nextcloud.com/tag/ncp) if relevant.
- Great for long form content
- Search the forum as most questions have already been asked and answered.
```
[Nextcloud](https://help.nextcloud.com/c/support/7) and [NCP](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm/33) Support in general, use [ncp tag](https://help.nextcloud.com/tag/ncp) if relevant.
- Great for long form content
- Search the forum as most questions have already been asked and answered.

*Minor editing for github section*

```
- [Bug Reports & Feature Requests](https://github.com/nextcloud/nextcloudpi/issues) for the NextcloudPi project.
  - Search `is:issues` to see what has already been discussed and closed.
- [Documentation](https://t.me/NCP_Wiki_Team) related requests.
```
- [Bug Reports & Feature Requests](https://github.com/nextcloud/nextcloudpi/issues) for the NextcloudPi project.
  - Search `is:issues` to see what has already been discussed and closed.
- [Documentation](https://t.me/NCP_Wiki_Team) related requests.

*Minor editing for telegram section*

```
[Nextcloud General Discussion](https://t.me/nextcloud) is on [Telegram](https://t.me/nextcloud) and Matrix.
[NextcloudPi discussion](https://t.me/nextcloudpi) is on Telegram.
[Documentation Team for ncp]([Documentation](https://t.me/NCP_Wiki_Team) is on Telegram.
  - Please submit documentation requests through Github.
```
[Nextcloud General Discussion](https://t.me/nextcloud) is on [Telegram](https://t.me/nextcloud) and Matrix.
[NextcloudPi discussion](https://t.me/nextcloudpi) is on Telegram.
[Documentation Team for ncp](https://t.me/NCP_Wiki_Team) is on Telegram.
- Please submit documentation requests through Github.

ovpc, 2020-10-31:
incorporated, thanks @sunjam

ovpc, 2020-11-01:
Reopening this, as we need to establish some guidelines as to: 
how messages on Telegram are going to be judged and how we are going to respond to them when they meet. 
Thanks for pointing out @theCalcaholic

sunjam, 2021-09-10:
Checking in on this...

Do guidelines need to be added to this particular document, or a different document specific to Telegram chat?

ovpc, 2021-09-10:
https://docs.nextcloudpi.com/en/forum-github-or-telegram-chat-which-to-use-when/
What guidelines would like to add?

closed 2021-09-10 by ovpc