nextcloud/nextcloudpi #1204: X-post Forum posts to Telegram chat

sunjam, 2020-10-30:
**Description**  
This is a feature request for auto-forwarding particular forum posts to the Telegram chat based on Tags (`ncp`) and/or Categories.  (`Appliances`).  Related to #1202 
[Would be assisted by enforcing one tag per appliance post on forum](https://help.nextcloud.com/t/please-enforce-use-of-tags-under-appliances/96734) category, to confirm whether post relates to ncp, Docker, VM or Snap appliance.

**Example**
An [ncp tag](https://help.nextcloud.com/tag/ncp) x-posts to the Telegram chat directly.  Follow the link and you'll see this currently equals about 2 - 3 posts per day.  Doesn't seem particularly spammy, so I think it would be most useful in encouraging use of the forum.  This x-posting can be further restricted or expanded by particular [categories](https://help.nextcloud.com/categories) on the forum.

Thanks for considering!

nachoparker, 2020-11-01:
good idea, might be noisy though, we're discussing using a moderator bot also

sunjam, 2020-11-23:
Here is my current layout concept.  Posts can be 1st Post Only or include replies (see screencap below)
#### Wiki Doc chat
- Appliances category with both `documentation` +`ncp` tags will post to wiki doc chat - **1st Post Only**
- How-To category with `ncp` tag posts to wiki doc chat as **1st Post Only**

#### ncp chat
- Specific mention of `@nextcloudpi` within a post will x-post to our general telegram chat. possibly 1st post only.
  - this means `@nextcloudpi` will exist the forum as a "group" which can be mentioned similar to the ncp tag.  This allows people to mention the project within posts as opposed to just topics.

Here is how rules are defined.
[image]

sunjam, 2021-09-11:
This is now all possible via RSS bots on matrix end and feeds such as https://help.nextcloud.com/tag/ncp.rss

sunjam, 2021-09-11:
Implemented.

sunjam, 2021-09-19:
RSS bot support is not posting to Telegram, just on Matrix side. re-opening.

sunjam, 2021-09-20:
Resolved by tweaking on Matrix side.  Thanks @theCalcaholic

closed 2021-09-20 by sunjam