nextcloud/nextcloudpi #1205: Update documentation for arm64

sunjam, 2020-11-04:
Documentation, particularly regarding Pi devices, centers around armhf.  Pi Foundation itself does not list arm64 or lite version on their website, but it is available.  New Nextcloud features (Collabora and Talk Matterbridge) are only focusing on arm64.

* Does this also mean re-writing the default curl script?  I believe that is armhf specific on some devices at least.
* Most of the documentation on the website needs to be re-written.
* There are differences between the older documentation on the website ownyourbits.com and docs.nextcloudpi.com/
* Supported devices also need to be updated, but that is likely a separate (though related) issue.

Larsene, 2020-11-05:
You can easely install Nextcloudpi on [PiOS Lite 64bit](https://downloads.raspberrypi.org/raspios_lite_arm64/images/). No change to the install with `# curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash` (no change). On a RaspberryPi4, this is highlyt recommended. more snappy, the ability to use CollaboraCODE, etc.

I hope most of the users use the curl script, and not the images, as none of them integrate a 64bit OS.

sunjam, 2020-12-05:
> I hope most of the users use the curl script, and not the images, as none of them (for raspberry cards) integrate a 64bit OS.

64-bit images are now available as of 20.0.2 and updated, refined documentation is still needed for the website and docs

nachoparker, 2021-10-15:
closing due to inactivity, reopen if you feel this is a mistake

nachoparker, 2021-10-15:
I think that this has been addressed already