nextcloud/nextcloudpi #1206: `Code` formatting not supported on docs.nextcloudpi.com

sunjam, 2020-11-04:
Wordpress theme is breaking `code formatting` on all documentation at docs.nextcloudpi.com

ovpc, 2020-11-05:
Problem is known, and has been for some time.
We need help with css of the current theme (it is beyond my skill set), or find an other one that supports the multi-language plugin, and has a similar kind of wiki or category listing as a landing page.
I have looked at and tried many other themes on a test server I maintain, on which the current one was build before taking the current one to production, but have not been able to find a free and suitable replacement theme.
Thanks for your help and suggestions.

sunjam, 2020-11-05:
dm'ed you.  Let me know on Wordpress configuration, or the issue can be reported directly to [BuddyDocs support forum](https://wordpress.org/support/plugin/buddypress-docs/).

ovpc, 2020-11-05:
We have Buddy Docs Version 1.0
It is hosted on shared hosing service from Nacho, same as ownyourbits. 
Pligins in use:
[image]

theCalcaholic, 2020-12-07:
@ovpc Couldn't you just enable markdown? If I read it correctly, it's built into wordpress itself, no plugin needed.

https://wordpress.com/support/wordpress-editor/blocks/markdown-block/#enabling-markdown

ovpc, 2020-12-08:
Thanks for tip, but I dont see any option, suggested by instructions, to enable markdown. (Settings>writing)
I have to install a plugin (Github MD), and that editor splits the screen in two, dont like it at all, and seems to make things even worse....

sunjam, 2020-12-10:
[Issue filed on BuddyPress](https://wordpress.org/support/topic/code-formatting-is-broken-on-our-buddypress/#new-topic-0)

sunjam, 2020-12-10:
Copypasta from support

"Are you using any plugin for formatting/hilighting the code. I can see
that the pre/code block is there. The theme will not apply any style if you
are using a pre tag.

I will recommend using a syntax hilighter to have better control."

ovpc, 2020-12-11:
"Pre/code block" Not that I know of....
Cab see Enabled plugins in screenshot above.

nachoparker, 2021-10-15:
closing due to inactivity, reopen if you feel this is a mistake

closed 2021-10-15 by nachoparker